hzeller/rpi-rgb-led-matrix #336: P10 1/2 scan multiple pixel lit

beikimajid, 2017-06-06:
Hi
I have P10 16x32 1/2 scan.When i run pixel-test give me this result :
I send (0,0)  then  (7,0) (7,1) (7,4) (7,5)  lit.

I use this pixel-mapping for 1/4 scan & it work:
int new_x = 0;
int new_y = 0;
if (y < 4 || (y > 7 && y < 12)) {
new_x = (x * -1) + (24 * ((x / 8) + 1)) - 17;
new_y = y;
} else {
new_x = x + (8 * ((x / 8) + 1));
new_y = y - 4;
}

I'm confused why several light goes on.Can you help me to write transformer for 1/2 scan?

Thanks a lot

beikimajid, 2017-06-08:
Who does not idea?

hzeller, 2017-06-08:
You did not adhere to the rule of bug reporting: add all relevant information :)

You have not really specified your panel (size ?). How does the connector on the back look like (Does it only have A, no B, C ? Because 1:2 scan would be something like this). Did you figure out the pixel mapping by going through all available pixels one by one ?

beikimajid, 2017-06-11:
Hi
As I said I use P10 16x32 1/2 scan.This is my module:
R1   G1
B2   GND
R2   G2   
B2   GND
A    B
NC  GND
CLK  LAT
OE GND

I've used the same module with 1/4 scan and it's OK.
But with 1/2 scan I can not figure out the pixel mapping by going through all available pixels one by one because as example when I send (0,0)  then (7,0) (7,1) (7,4) (7,5) together lit.

beikimajid, 2017-06-15:
Does anyone have an idea?

hzeller, 2017-06-15:
It is not clear why you want to use 1/2 scan (which would mean you would only use address `A`).

If multiple rows are lit, maybe you got a different kind of panel, that does some internal address advancing ?

closed 2017-07-30 by hzeller