hzeller/rpi-rgb-led-matrix #337: Permission denied

maxwellflitton, 2017-06-06:
I have an adahat and a 32x32 med matrix. I have followed the instructions and the responses on the terminal have been the same as in the tutorial. However, I get to running the first example: matrixtest.py. If I run it through the idel nothing happens and no errors are thrown up. If I run it through the terminal I get the following:

can't open /dev/mem: : Permission denied
Attempt to init outputs but initialized.
Result: 0x0 v 0xCF33070
python: framebuffer.cc:97: static void rgb_matrix::RGBMatrix::Framebuffer::InitGPIO(rgb_matrix::GPIO*): Assertion `result == b.raw' failed.
Aborted

What does this mean? Is my soldering off? I have no crossed wires

hzeller, 2017-06-06:
you need to run things as root as the library needs access to underlying hardware. Prepend `sudo` before your command.

maxwellflitton, 2017-06-06:
wow thanks for the quick reply. It now works showing the following:

Result: 0xCF33070 v 0xCF33070

But nothing shows up on the board. Is that result expected? (Sorry for being slow at this)

hzeller, 2017-06-06:
First off: do not use the Adafruit tutorial, it is hopelessly outdated and only works if you use their ancient fork of my library. Unfortunately, they don't bother to update it. If you use this library, please follow the READMEs here.

Not sure what the 0xCF33070 is about.

maxwellflitton, 2017-06-07:
Thanks, I have downloaded your library and read the documentation. I am still having some issues. If I try and run the sample scripts it says that there is no module name 'rgbmatrix.core'. 

I tried installing it via pip and I got the following:

Command /usr/bin/git clone -q
https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/python.git
/tmp/pip-l2C5dT-build failed with error code 128 in None
Storing debug log for failure in /home/pi/.pip/pip.log

And if I just ran the: sudo python matrixtest.py I would get the following:
Result: 0xCF33070 v 0xCF33070

I have a adafruit hat so I also ran the following: 
sudo ./runtext.py --led-gpio-mapping=adafruit-hat

which gave the following:
sudo ./runtext.py: command not found 

How do I install the 'rgbmatrix.core' so I can import it in a script? Sorry if the answer is clear, I appreciate that you're busy, I've tried multiple approaches and nothing seems to be working.

hzeller, 2017-06-08:
Please read the instructions.

You need to check out the full rpi-rgb-led-matrix, not only the python subdirectory.

Then you go into the Python directory and follow the install and build instructions
```
$ make
$ sudo make install
```

The `runtext.py` is in the sample directory.

closed 2017-07-30 by hzeller