hzeller/rpi-rgb-led-matrix #338: 32x16 rgb led notworking

zishant, 2017-06-08:
I am using hzeller library but my pannel is not working well. it display some random pattern

hzeller, 2017-06-08:
Did you read the [Troubleshooting section](https://github.com/hzeller/rpi-rgb-led-matrix#troubleshooting) and followed all advice there ?
After you have worked with all the advice there, what is not working ?

zishant, 2017-06-08:
i turn off on board sound (dtparam=audio=off).but still not working.
in my led panel there NC pin instead pin C,i used NC as a pin C it would create any issue.
i have connect led panel manually to raspi using female to female pi.

zishant, 2017-06-08:
[image]

hzeller, 2017-06-08:
NC means "not connected", so connecting C to it has no effect. This could be a 1:4 multiplexing panel.
How big is your display ? Width x height ?

zishant, 2017-06-08:
32x16
[image]

hzeller, 2017-06-08:
Ok, so you have 1:4 multiplexing.
You need to give `--led-rows=8 --led-chain=2` for this one, then write a transformer that maps all the pixels to the right place. You can do that as little exercise yourself or look at  #313 where someone wrote such a transformer for their board.

zishant, 2017-06-08:
thanks ok sir i will try and let you know the result...
so should i left the NC pin as it is or connect to pin C

hzeller, 2017-06-08:
Don't connect it. It is not needed.

zishant, 2017-06-08:
i am not getting where should i change the code...pleas help me out i am stuck from last four days.

zishant, 2017-06-08:
shall i change in transformer.cc file and also which function...
i am little slow in c++ please help me out...

hzeller, 2017-06-08:
yes, the simplest is if you copy a transformer and give it a different name. Then put in the code from #313 . You then can use this transformer in your code by using the ApplyStaticTransformer() call.

zishant, 2017-06-08:
Actually I am using library so it's difficult to used my own transformation.cc 
Please let me know simple way...
Let me know which function do I need to change in your transformation.cc

zishant, 2017-06-08:
sir please give solution sir i need your help....asap
i changed 

void UArrangementTransformer::TransformCanvas::SetPixel(
  int x, int y, uint8_t red, uint8_t green, uint8_t blue) {
  if (x < 0 || x >= width_ || y < 0 || y >= height_) return;
  const int slab_height = 2*panel_height_;   // one folded u-shape
  const int base_y = (y / slab_height) * panel_height_;
  y %= slab_height;
  if (y < panel_height_) {
    x += delegatee_->width() / 2;
  } else {
    x = width_ - x - 1;
    y = slab_height - y - 1;
  }
  delegatee_->SetPixel(x, base_y + y, red, green, blue);
}

with new one
void UArrangementTransformer::TransformCanvas::SetPixel(int x, int y, uint8_t red, uint8_t green, uint8_t blue) {
int new_x = 0;
int new_y = 0;
if (y < 4 || (y > 7 && y < 12)) {
new_x = (x * -1) + (24 * ((x / 8) + 1)) - 17;
new_y = y;
} else {
new_x = x + (8 * ((x / 8) + 1));
new_y = y - 4;
}
delegatee_->SetPixel(new_x, new_y, red, green, blue);

i haven't check yet

hzeller, 2017-06-08:
If you use the library, you still can write your own transformer in your code, no problem. My suggestion is to just copy a sample implementation from the transformer.cc into your own code and modify that which might be simpler than doing it from scratch.
The interface is public and you can write your own implementation and pass it to the public ApplyStaticTransformer().

zishant, 2017-06-08:
I change void UArrangementTransformer::TransformCanvas::SetPixel function with my function which is from #313 still not working but change the behavior little bit.i think we need snake

zishant, 2017-06-08:
Transformer

zishant, 2017-06-09:
Hi hzeller,

The problem is solved.. but display gets hanged I don't know why

zishant, 2017-06-09:
where should i add this.. matrix->ApplyStaticTransformer(rgb_matrix::P10outdoorTransformer());

 i add like this...
RGBMatrix *matrix = CreateMatrixFromOptions(matrix_options, runtime_opt);
matrix->ApplyStaticTransformer(rgb_matrix::P10outdoorTransformer());

but still display hanged

hzeller, 2017-06-09:
What do you mean with 'hanged' ?

zishant, 2017-06-09:
Suddenly it's stop displaying and display random patterns

Where should I call this ApplyStaticTransformer(rgb_matrix::P10outdoorTransformer()); in demo-main.cc

hzeller, 2017-06-09:
Did you read the troubleshooting section ? Did you follow the advice in removing the 1-wire configuration ? Are you running Raspbian-Lite ?

zishant, 2017-06-09:
Oh I haven't remove I will..
It's just rasbian simple one

zishant, 2017-06-10:
Thanks a lot Hzeller,
Now its working fine...without and issue.
but i want to display same message using two 32x16 panel parallel i.e 32x32..

zishant, 2017-06-10:
Text-example is not working...for 1/4 scanning  
Where should I change in text-example.cc for 1/4 scanning to be work....

hzeller, 2017-06-11:
You need to set your transformer of course to the RGBMatrix.

mbates14, 2017-07-27:
I noticed you support HDMI siphoning with crop or scale, How do you apply this transformer to that? Also does this knock down the tiles from 12 to 6? or can a Pi3 still run the 12 with this transformer?

zishant, 2017-07-28:
I haven't used HDMI bro..

mbates14, 2017-07-28:
I should have been more clear when I said "you" in my post, I was referring to the Author of the project/library. Sorry.

hzeller, 2017-07-28:
@mbates14 make sure to not hijack non-related issues for your question to avoid confusion.

This library does not do anything with HDMI, and you can't capture HDMI directly. But someone built a framebuffer capture which then shows on a RGB matrix. Check out https://github.com/adafruit/rpi-fb-matrix

closed 2017-07-28 by hzeller

zishant, 2017-07-28:
It's ok man....it's happen sometime