hzeller/rpi-rgb-led-matrix #339: Led-Image-Viewer update is not supporting .png .gif

primanator, 2017-06-08:
With the recent update the led-image-viewer gives "*** not a usable image format" when trying to display any .png or .gif.
Before update everything was running ok, checked it by downloading and compiling the december repository version.

hzeller, 2017-06-08:
This is odd. Can you give a command line you are using, an example image and the version numbers of the graphics-magic libraries ?
Is there any other error message that is shown ?

hzeller, 2017-06-09:
I've now added more diagnostic output, maybe it helps to figure out what is happening in your case.
I suppose you recompiled everything from scratch, with a nice `make clean` in the toplevel directory and then a `make` in the utils directory just in case there is something funny with the linking ?

hzeller, 2017-06-11:
If it still doesn't work - do you see anything in the diagnostic output ?

hzeller, 2017-06-20:
No response. Assuming it started working.

closed 2017-06-20 by hzeller