hzeller/rpi-rgb-led-matrix #341: Implementing "S" arrangement transformation

chrisoutwright, 2017-06-12:
As the adapter for 3 chains will take some time to ship and mount, I thought about going about the implementation for 9x 32 panel 1-chain "S" square-arrangement, 
so one additional row over the UArrangementTransformer.

As somebody tried it already in this issue [post](https://github.com/hzeller/rpi-rgb-led-matrix/issues/307), I tried to replicate the issue with it. I posted there a comment as to what is amiss in the display.

My question would be if it is only the `void SArrangementTransformer::TransformCanvas::SetPixel(...)` method needing a fix, or is it more than this in @soyxan's code?

I would be extremely glad if somebody has an idea about it, or even a solution.

chrisoutwright, 2017-06-12:
Nearly got there for 3 rows with 3 panels, only the last row is now incrementing lines for characters in a wrong way:

```
void SArrangementTransformer::TransformCanvas::SetPixel(
  int x, int y, uint8_t red, uint8_t green, uint8_t blue) {
  if (x < 0 || x >= width_ || y < 0 || y >= height_) return;
  const int slab_height = virtualrows_*panel_height_;   // one folded s-shape
  const int base_y = (y / slab_height) * panel_height_;
  y %= slab_height;
   if (y < panel_height_) {
   x += delegatee_->width()*2/3 ;

     }
 if ((y < 2*panel_height_ ) && (y >= panel_height_)) {
    x = 2*width_ - x - 1;
    y = slab_height *2/3 - y - 1;
  }

 if  ((y < 3*panel_height_ ) && (y >=2* panel_height_)) {
    x += delegatee_->width()*0;
    y = slab_height  - y - 1;

  }
  delegatee_->SetPixel(x, base_y + y, red, green, blue);
```
}

chrisoutwright, 2017-06-13:
For those interested, the now **_correctly working version_** for the relevant method is:

```
void SArrangementTransformer::TransformCanvas::SetPixel(
  int x, int y, uint8_t red, uint8_t green, uint8_t blue) {
  if (x < 0 || x >= width_ || y < 0 || y >= height_) return;
  const int slab_height = virtualrows_*panel_height_;   // one folded s-shape
  const int base_y = (y / slab_height) * panel_height_;
  y %= slab_height;
   if (y < panel_height_) {
   x += delegatee_->width()*2/3 ;

     }
 if ((y < 2*panel_height_ ) && (y >= panel_height_)) {
    x = 2*width_ - x - 1;
    y = slab_height *2/3 - y - 1;

  }

 if  ((y < 3*panel_height_ ) && (y >=2* panel_height_)) {
    // x = width_ - x - 1;
     y =  y - slab_height*2/3 ;

  }
  delegatee_->SetPixel(x, base_y + y, red, green, blue);
}
```

closed 2017-06-13 by chrisoutwright

soyxan, 2017-10-07:
I am trying to use your code for my panel configuration of 3 rows of 10 panels (32x32), with no luck!

Have you modified the SArrangementTransformer::TransformCanvas::SetDelegatee() function? What I have to modify in your code to get it working?

Thanks in advance!

chrisoutwright, 2017-10-10:
Hi soyxan,
Are you doing 30 panels in series or 3 parallel rows of 10 each?
The code I submitted is intended for 9 panels in square formation on one output line.
Concerning` parallel=1`, the code should work for 9 panels, as my pictures show.
One could scale the code for, say, 16 panels, but this would stress the single output line.
Doing anything other than square formation with higher than two panel rows (therefore S-Transformer) would require testing and adapting the code to see how to map the pixels.
What do you think?