hzeller/rpi-rgb-led-matrix #345: RGB LED Matrix HAT

cbrancato96, 2017-06-14:
I've been having a lot of problems with the matrix HAT. Everything seems soldered and connected appropriately and I tried the suggestions on the readme yet when I run any of the tests the matrix gives a random display. When I play/fiddle with the IDC connection to the matrix or the power/gnd connection the matrix changes its display randomly.

hzeller, 2017-06-14:
As always when reporting an issue it is important to include the details.

what HAT are you talking about? There are several hardware adapters Pi -> Matrix out there.
What flags are you giving the binary?
How many displays chained?
what size display?

cbrancato96, 2017-06-14:
Thanks for the help.
RGB LED Matrix Adafruit HAT
the flag that i am giving is --led-gpio-mapping=adafruit-hat
I am using one 32 x 32 display

hzeller, 2017-06-20:
Ok, so you made sure to disable 1 wire communication which seems to be enabled on some Raspbian (though if you followed the tips you have Raspbian lite which did not exhibit this configuration issue last time I checked).

If you still get random output, maybe you got a display that has a differnent interface ? How does the connector look like ? Usually, there is some printing about the signals it expects.

cbrancato96, 2017-06-20:
Fixed it by not using the hats hub75 interface, will try with the hat again soon tho.

cbrancato96, 2017-06-21:
Can you provide another example to write text to the display?

hzeller, 2017-06-21:
There is an [self-explanatory API](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/include/graphics.h#L82) and a [C++ example](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/examples-api-use/text-example.cc) already. Even Python examples.

What is the question that these don't answer yet ?

cc96cc, 2017-06-21:
when i try to rewrite the c++ example it seems i can never load the font in correctly

hzeller, 2017-06-21:
Can you give a command line you are using ?

cc96cc, 2017-06-21:
not command line, just trying to use DrawText correctly
rgb_matrix::DrawText(matrix, font, x, y + font.baseline(), color, &bg_color, line);

hzeller, 2017-06-21:
You said you have trouble loading the font, but you did not show how you load the font.

cc96cc, 2017-06-21:
rgb_matrix::Font font";

if (!font.LoadFont("/home/pi/rpi-rgb-led-matrix/fonts/6x10.bdf")) {
   std::cout << "Couldn't load font" << std::endl;
    return 0;
  }

hzeller, 2017-06-21:
You really need to provide the necessary information here. Does it return false there ? Or does it continue, but the font is not usable ? Is the file readable ?
If you need help, you need to provide everyone the relevant information to be able to help.

closed 2017-07-30 by hzeller