hzeller/rpi-rgb-led-matrix #346: Only one row of leds light up

grant100, 2017-06-23:
I have a 32x32 rgb matrix and an a pi3, and I followed the wiring readme and connected everything up through a breadboard. When I run the demos (any of them) only one row lights up and changes colors, all the others are blank. I tried using the rotation flag to see if it would change and it didn't.  The rgb matrix is powered by a 5v 4A supply. I don't have a level shifter, I'm just plugging everything in. Any ideas as to why I only get one of 32 rows?

hzeller, 2017-06-23:
a single row sounds interesting. If the address lines would not work (which could well be because of no level shifter), it would still be two lines that would do something (as the first line is the R0,G0, B0 and the second row gets R1, G1, B1), I think the next step would be to add a level shifter (probably easier to solder one up than doing it through the breadboard https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3 )

grant100, 2017-06-30:
I added a level shifter and it is still only producing 1 row. This makes me think the matrix itself has a problem. I've got a new one on the way, I'll let you know what I find.

hzeller, 2017-07-30:
Closing this for now. If you have an update with the new panel, just comment here.

closed 2017-07-30 by hzeller