hzeller/rpi-rgb-led-matrix #347: i2c with adafruit hat

dmelladoc, 2017-06-23:
I'm using your library with the AdaFruit HAT for some experiments and everything works well, but when I try to integrate i2c communication with another device and using the library, my rPi can't connect to the device. 
Both i2c and the library works fine when used separated, and I don't seem to find any issues with the code, or anything.

Does the library use the SDA and SCL pins from the rPi while communicating using the config for the HAT? if so, does using your recommended designs solve this?

thanks in advance!

hzeller, 2017-06-24:
If you use the Adafruit HAT hardware mapping, the pins that are occupied by the SDA and SCL are not being used:
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/hardware-mapping.c#L73

Having said that, the led matrix constantly sends out data on the GPIO pins that are configured for output, so if SDA/SCL is done via GPIO not via a different PinMux, this could influence that. I have never used I2C on the Pi, so I don't really know the details.

Also it could just be stray noise from the rapid switching on adjacent lines. Have you properly terminated the I2C bus with sufficiently low resistors ?

dmelladoc, 2017-07-03:
After almost 2 weeks trying everything, including your idea, and monitoring all outputs with a oscilloscope, I found that you have to initialize i2c BEFORE initializing a led matrix canvas in order for both things to work.

Thanks for the advice, though. I noticed some issues with my i2c setup with your suggestion.

closed 2017-07-17 by dmelladoc