hzeller/rpi-rgb-led-matrix #348: Glitches on the matrix

tychop, 2017-07-06:
Has anybody got some tips to solve these glitches?

[image]

My power supply is 5v, 10Amps

And the command I ran was: `sudo ./demo --led-rows=32 --led-chain=2 --led-slowdown-gpio=3 --led-pwm-lsb-nanoseconds=1500 --led-slowdown-gpio=3 --led-no-hardware-pulse -D1 runtext16.ppm `

I think I tried most of the command parameters, but I keep seeing the glitches....

maddogs, 2017-07-06:
Do you have anything that could be interfering in the background such as PiGPIO or 1 wire sensor

hzeller, 2017-07-06:
Please read the troubleshooting section. This looks like you haven't disabled the 1 wire sensor. Also, do not use PiGPIO.

closed 2017-07-06 by hzeller

tychop, 2017-07-06:
No, nothing.. Even the sound is turned off in /boot/config.txt.

tychop, 2017-07-06:
@hzeller I understand it's not an issue with the library... Where do you think I should ask this question?

hzeller, 2017-07-06:
The glitches you show look like the 1-wire sensor.
Also make sure to run Raspbian Lite as the regular Raspbian packs all kinds of things on there that might interfere with your GPIO.

tychop, 2017-07-06:
Thnx!, I'll go and get the raspbian light. That sounds like it could help.

tychop, 2017-07-06:
Awesome. Installed the lite version and all is peachy with just the nr of rows & chain.
👍🏻