hzeller/rpi-rgb-led-matrix #350: C++ Performance questions with "text-example.cc"

chrisoutwright, 2017-07-11:
By building a slight different version of the original code from text-example.cc,
I am testing time/date output on the LED RGB display. 

The code also deviates by the original by a "word-counting" through an added @-separator token finder, so that color can be changed word-wise, while these colors are given via flags in the options.

Displaying static text in an 2 or higher seconds update interval (using `sleep(2)`) is fine,
when I include a changing string, via call of the time_o() in   ´strncpy(temp2, time_o(), 26);´ :

		#include "led-matrix.h"
		#include "graphics.h"

		#include <getopt.h>
		#include <stdio.h>
		#include <stdlib.h>
		#include <string.h>
		#include <unistd.h>
		#include <time.h>
		#include <sys/time.h>
		#include <ctime>
		#include <stdlib.h>

		using namespace rgb_matrix;

		static int usage(const char *progname) {
		  fprintf(stderr, "usage: %s [options]\n", progname);
		  fprintf(stderr, "Reads text from stdin and displays it. "
				  "Empty string: clear screen\n");
		  fprintf(stderr, "Options:\n");
		  rgb_matrix::PrintMatrixFlags(stderr);
		  fprintf(stderr,
				  "\t-f <font-file>    : Use given font.\n"
				  "\t-b <brightness>   : Sets brightness percent. Default: 100.\n"
				  "\t-x <x-origin>     : X-Origin of displaying text (Default: 0)\n"
				  "\t-y <y-origin>     : Y-Origin of displaying text (Default: 0)\n"
				  "\t-C <r,g,b>        : Color. Default 255,255,0\n");
		  return 1;
		}

		static bool parseColor(Color *c, const char *str) {
		  return sscanf(str, "%hhu,%hhu,%hhu", &c->r, &c->g, &c->b) == 3;
		}

		char*  time_o()
		{
			time_t timer;
			char buffer2[26];
			struct tm* tm_info;
			
			time(&timer);
			tm_info = localtime(&timer);
			strftime(buffer2, 26, "%H:%M:%S", tm_info);
			
			return buffer2;
		 
		}

		int main(int argc, char *argv[]) {
		  RGBMatrix *canvas = rgb_matrix::CreateMatrixFromFlags(&argc, &argv);

			char temp2[26];   
			char* whole[] = {temp2,"?d","MB@CH","World","BBys"};
			char s_remote_port[6];
			int remote_port = sizeof(whole)/sizeof(*whole);
			
		   // snprintf(s_remote_port, sizeof(s_remote_port), "%d", remote_port);
			

		  Color black(0,0,0);
		  Color color(255, 255, 0);
		  Color color0(255, 255, 0);
		  Color color1(255, 255, 0);
		  Color color2(255, 255, 0);
		  Color color3(255, 255, 0);
		  Color color4(255, 255, 0);

		  const char *bdf_font_file = NULL;
		  int x_orig = 0;
		  int y_orig = -1;
		  int brightness = 100;

		  
		  // Colors q-1,w-2,e-3,r-4
		  
		  int opt;
		  while ((opt = getopt(argc, argv, "x:y:f:C:b:q:w:e:r:")) != -1) {
			switch (opt) {
			case 'b': brightness = atoi(optarg); break;
			case 'x': x_orig = atoi(optarg); break;
			case 'y': y_orig = atoi(optarg); break;
			case 'f': bdf_font_file = strdup(optarg); break;
			case 'C':
			  if (!parseColor(&color, optarg)) {
				fprintf(stderr, "Invalid color spec.\n");
				return usage(argv[0]);
			  }
			  break;
			case 'q':
			  if (!parseColor(&color1, optarg)) {
				fprintf(stderr, "Invalid color spec.\n");
				return usage(argv[0]);
			  }
			  break;
			case 'w':
			  if (!parseColor(&color2, optarg)) {
				fprintf(stderr, "Invalid color spec.\n");
				return usage(argv[0]);
			  }
			  break;
			case 'e':
			  if (!parseColor(&color3, optarg)) {
				fprintf(stderr, "Invalid color spec.\n");
				return usage(argv[0]);
			  }
			  break;
			  case 'r':
			  if (!parseColor(&color4, optarg)) {
				fprintf(stderr, "Invalid color spec.\n");
				return usage(argv[0]);
			  }
			  break;

			
			default:
			  return usage(argv[0]);
			}
		  }

		  if (bdf_font_file == NULL) {
			fprintf(stderr, "Need to specify BDF font-file with -f\n");
			return usage(argv[0]);
		  }

		  if (canvas == NULL)
			return 1;

		  /*
		   * Load font. This needs to be a filename with a bdf bitmap font.
		   */
		  rgb_matrix::Font font;
		  if (!font.LoadFont(bdf_font_file)) {
			fprintf(stderr, "Couldn't load font '%s'\n", bdf_font_file);
			return usage(argv[0]);
		  }

		  if (brightness < 1 || brightness > 100) {
			fprintf(stderr, "Brightness is outside usable range.\n");
			return 1;
		  }

		  canvas->SetBrightness(brightness);
		   
		  bool all_extreme_colors = brightness == 100;
		  all_extreme_colors &= color.r == 0 || color.r == 255;
		  all_extreme_colors &= color.g == 0 || color.g == 255;
		  all_extreme_colors &= color.b == 0 || color.b == 255;
		  if (all_extreme_colors)
			canvas->SetPWMBits(1);

		  const int x = x_orig;
		  int y = y_orig;
		  
		  color0 = color;
		  int xlan = 0;
		  char * pch;
		  int a = 0;
		  
		  strncpy(temp2, time_o(), 26);

		  //Runs once to display time and static strings
		  
		  while (a<=remote_port-1) {
			  
			  
			if ((y + font.height() > canvas->height())) {
				canvas->Clear();
				y = y_orig;
			 } 

			if (y<font.height()) {
				
					strncpy(temp2, time_o(), 26);

			 } 
			
			
			char temp[20];
			strncpy(temp, whole[a], 20);
			

			int wort = 0;
				for (pch = strtok(temp,"@");pch != NULL;wort++)
				{
				switch (wort) {
					case 0: color = color0;break;
					case 1: color = color1;break;
					case 2: color = color2;break;
					case 3: color = color3;break;
					case 4: color = color4;break;
				}
				
				if (pch[0]=='+') {
					color = color4;
				}
					
					
			

			

				rgb_matrix::DrawText(canvas, font, x + xlan, y + font.baseline(), color, pch);

				xlan = xlan+(strlen(pch)) * font.width();
				pch = strtok (NULL, "@");
				
				}

			xlan = 0;
			wort = 0;
			
			
			y += font.height();

			if (a==remote_port-1) {
				strncpy(temp2, time_o(), 26);	
				a = -1;
				sleep(2);

				}

			a=a+1;
			
			
		}	

		  // Finished. Shut down the RGB matrix.
		  canvas->Clear();
		  delete canvas;

		return 0;
		 
		}

I am getting a slight horizontal flicker on the display in the rows of the otherwise unchanging output.

Admitted, this flicker is much less than when I call the original text-example.cc in a while loop with piped output of a python script, but it would be great if one could reduce it further still.

I am using 4 64 Panels with options:
` --led-rows 64 --led-chain 4  --led-slowdown-gpio=2 --led-swap-green-blue --led-show-refresh --led-pwm-lsb-nanoseconds 130  --led-parallel=2 --led-pwm-bits=4 -f fonts/10x20.bdf`

One solution so far was to reduce the updating of the whole panel to the  infinitely updated row only, displaying the time, by reducing the first while to a single run and adding a separate while loop, like this:

			
			char temp3[26];
			int x2 = x_orig;
			int y2 = y_orig;
			while (true) {
				strncpy(temp3, temp2, 26);
				strncpy(temp2, time_o(), 26);

				rgb_matrix::DrawText(canvas, font, x2 , y2 + font.baseline(), black,temp3);
				rgb_matrix::DrawText(canvas, font, x2 , y2 + font.baseline(), color0,temp2);
				sleep(1);

				}

This seems to make possible an update interval of 1-sec. 
But is could mean, that when canvas->Clear(); is called this could have resulted in the more prominent flicker (than when simply overwriting the dynamic row with black font), or am I wrong?

hzeller, 2017-07-11:
You really want to write in an offscreen canvas and then `SwapOnVSync()` that. Otherwise, you are changing pixels while they are being pushed out to the screen, and that will cause glitches.

(Independent seen while skimming through the code: your `time_o()` function is faulty: you are return a pointer to a temporary buffer on the stack. Also, you really should adopt a coherent indentation, it is super hard to read this way).

chrisoutwright, 2017-07-11:
Thank you for the prompt reply. 
I'll try to fix the identation with another editor.
I  have thought about manipulating the canvas offscreen frist, but wasn't sure with what methods. I will try to build it in. 
About time_o(), I am not sure what the fault will entail as a consequence, as the output seemed unaffected. What would be the correct way to avoid this?

hzeller, 2017-07-11:
For indentation: I suggest just using emacs in tramp mode. That way, you can remotely edit something comfortably on your 'big' machine.

For `CreateFrameCanvas()` and `SwapOnVSync()` look at the numerous other examples here and in the util directory. Also [the documentation in the header file](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/include/led-matrix.h#L199). Essentially you create an offscreen canvas before first, then in your loop, you write everything on that offscreen canvas. The `offscreen = matrix->SwapOnVSync(offscreen);` will replace the current canvas with the offscreen one and will return the current canvas, which is now your offscreen one.
You don't need to define a class or anything.

`time_o()` might work by accident as the stack is not overridden yet but it is invalid code as you refer to a deleted part of memory. You typically solve this by passing the buffer you want to fill the time into as a parameter. In particular in your case, you anyway then strcpy() the result in a local buffer - passing the buffer directly makes it faster (and correct).

chrisoutwright, 2017-07-11:
Thanks very much for the link and explanation. I got it to work now perfectly!
I created the object `FrameCanvas *offscreen_canvas = canvas->CreateFrameCanvas();` 
right before the loop and replaced `canvas` with `offscreen_canvas` when it draws on the graphics library (DrawText).

Putting `offscreen_canvas = canvas->SwapOnVSync(offscreen_canvas);` right before sleep(), enabled me to produce a great flickerfree output!

hzeller, 2017-07-11:
Cool! Enjoy the flicker-free display :)

closed 2017-07-11 by hzeller