hzeller/rpi-rgb-led-matrix #353: Running Python codes

Barnabe22, 2017-07-14:
Hello. I'am a beginner and I want to run the python codes of this project. I have a 32x16 RGBmatrix LED panel. But I don't have the HAT for the matrix and i've done the standard wiring by connecting my rapsberry pi 3 pins to the HUB 75 pins. All is right until here. But the problem is when I run for example:
sudo python runtext.py
Nothing appears on the rgbmatrix LED panel. I don't know why. Will someone clarify something to me please ?

hzeller, 2017-07-14:
Did you try the demo program in the `examples-api-use/` directory first before you jump to Python ?

Barnabe22, 2017-07-14:
Yes that was the first thing I tried to do but it doesn't work.

hzeller, 2017-07-14:
Well, then  fix your wiring first. If you have directly connected the wires, you might be out of luck as some panels require 5V logic signals and the Pi only does 3.3V.
Read the troubleshooting section.
I suggest to just solder up an [level converting adapter](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3)

Barnabe22, 2017-07-14:
Right. I forgot to mention that I have used a system of M74HC125 integrated circuits as level converting adapter. I checked each gate with 3.3V as input and they output the 5V correctly

hzeller, 2017-07-14:
The something else is wrong. Follow each trace, make sure Power and GND are properly connected etc. If this is a standard RGB-Matrix, it should work directly with the standard wiring described here

https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md

(Hopefully you didn't follow some Adafruit tutorial, as that is several years outdated and they don't bother to update).

Barnabe22, 2017-07-14:
Ok thank for response. I'll check about. 
Another question 
It seems that the library works for 32x32 panels by default. 
If I want scroll texts on a 32x16 panels, which lines in the code will I modify in the runtext.py?

hzeller, 2017-07-30:
You just need to give the option `--led-rows=16` to the program.

closed 2017-07-30 by hzeller

Barnabe22, 2017-08-02:
Hello. I have checked and I see something appears on the panel but I have another problem. When I run the demo all LED in the top half of the LED panel are in blue color. And the bottom half ones show something scrolling but not clear at all.

hzeller, 2017-08-02:
Double check your wiring and that you gave the right options.

Barnabe22, 2017-08-02:
Ok it is done. The demo works fine now. But the python codes still not working.