hzeller/rpi-rgb-led-matrix #354: we chained many row

beikimajid, 2017-07-15:
Hi
I've linked 10 rows of  32x64 panels on a Raspberry Pi 3 with the "S arrangement" like this :

Raspberry

P01 >> p02
p04 << p03
p05 >> p06
p08 << p07
p09 >> p10
p12 << p11
.
.
p20 << p19

everything all right but FIX image flickers a little.
The POWER IS OK and I use --led-pwm-lsb-nanoseconds , --led-slowdown-gpio ,--led-pwm-bits and other parameters  BUT I did not get any results and Still flickering very little and probably for more panel Will be more.
**I do not want to use parallel parameter and active3 borad.**

What number of Module connected in series(without parallel) does Raspberrypi3 support ?
How to use two Raspberry Pi's and have them synchronize a framebuffer via UDP packets?
Is there any other way?

hzeller, 2017-07-15:
You can see with the `--led-show-refresh` the refresh rate that it can do in the current configuration.

20 panels in series is a lot; so there is a lot of time spent in just clocking the data in. Not only will it flicker, the panels will look a bit darker also. I guess if you want to get rid of the flicker, you really should use parallel chains and an active3 board. If you set chain length to 7, you'd see how the refresh rate changes if you used parallel chains (because 7x3=21, which covers the 20 panels).

The number of series panels is indefinite, but of course, the refresh rate will suffer as you see. Also it might be limited by some RGB matrix that accumulate timing issues on the clock line in long sequences.

A simple way is indeed to send different parts of the image to multiple Pis and have them display synchronized content.

beikimajid, 2017-07-22:
my refresh rate is 46Hz to 61Hz. How to config?

hzeller, 2017-07-22:
with 20 panels connected, things are naturarlly a bit slower, but with `--led-chain=20 --led-slowdown-gpio=2` you should still get about 100Hz refresh rate with 11 bit PWM and > 300 Hz with 1 bit PWM (if you only need a few colors).

Please read the README, it contains a description of all the config parameters that influence the refresh rate.

closed 2017-07-30 by hzeller