hzeller/rpi-rgb-led-matrix #355: python directory not downloaded

wagadu, 2017-07-15:
Hi there
I am new to GitHub but got to run the demos with the adafruit board and raspberry pi. However, I'd like to use the python subdirectory. Unfortunately its not downloaded and I don't get it to download. Any idea why its not sync even I went according to your tutorial? Thank you!

maddogs, 2017-07-15:
So in the root directory of  \rpi-rgb-led-matrix\

You've tried to run this command:

`sudo apt-get update && sudo apt-get install python2.7-dev python-pillow -y`

And it's failed? If it's something else, can you list the steps of what you've tried.

wagadu, 2017-07-15:
Thanks for your quick reply. Yes I've tried that. It fails saying its already installed:
~/rpi-rgb-led-matrix-master $ sudo apt-get update && sudo apt-get install python2.7-dev python-pillow -y
Hit http://mirrordirector.raspbian.org jessie InRelease                         
Hit http://mirrordirector.raspbian.org jessie/main armhf Packages                  
Hit http://mirrordirector.raspbian.org jessie/contrib armhf Packages                           
Hit http://mirrordirector.raspbian.org jessie/non-free armhf Packages                          
Hit http://mirrordirector.raspbian.org jessie/rpi armhf Packages                               
Hit http://archive.raspberrypi.org jessie InRelease                                            
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en_GB  
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en
Ign http://mirrordirector.raspbian.org jessie/main Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/main Translation-en
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en
Hit http://archive.raspberrypi.org jessie/main armhf Packages
Hit http://archive.raspberrypi.org jessie/ui armhf Packages
Ign http://archive.raspberrypi.org jessie/main Translation-en_GB
Ign http://archive.raspberrypi.org jessie/main Translation-en
Ign http://archive.raspberrypi.org jessie/ui Translation-en_GB
Ign http://archive.raspberrypi.org jessie/ui Translation-en
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'python-pil' instead of 'python-pillow'
python-pil is already the newest version.
python2.7-dev is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 98 not upgraded.

maddogs, 2017-07-15:
okay, so you've then run the next two commands?

```
make build-python
sudo make install-python
```

Then you've cd'd to the python samples directory and tried to run a test?

wagadu, 2017-07-15:
Unfortunately I get the error:
make build-python
make: *** No rule to make target 'build-python'.  Stop.

maddogs, 2017-07-15:
Starting from the beginning, did you run a command like this to get the Library on your Pi

`git clone https://github.com/hzeller/rpi-rgb-led-matrix.git`

wagadu, 2017-07-15:
So sorry. That's it. I took a zip from ada fruit website instead of the git clone ...
Have a great day!

hzeller, 2017-07-15:
Thanks @maddogs for chiming in and helping!
Looks like things are working well now.

closed 2017-07-15 by hzeller