hzeller/rpi-rgb-led-matrix #356: 16x64 Red Panel

Szili89, 2017-07-15:
Dear Henner Zeller!

In my case I am trying to operate a 16x64 LED panel which has a HUB08 connector. I have managed to connect the cables and start some demos but I could not set up properly the program.

When I write -r 16 -c 2 than the picture is in the middle in the my panel but the half of the screen do not lit up. When I write -r32 -c2 instead, than the full screen is working but the half of the picture is missing.

Which part of the program should I check to be able to fix this issue?

I have enclosed two pictures of the rotating-block-generator.py sample.
Best regards,
Szilard

[image]
[image]

hzeller, 2017-07-15:
I wonder if this only has one half panel. After all, the panel is 16 high, and it has A, B, C, D. So I suspect R2 and G2 don't have any function.

Try to set the `-DONLY_SINGLE_SUB_PANEL` define while compiling and see if this helps in this case:
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/Makefile#L117

Szili89, 2017-07-16:
Yes you have right. This panel arrived with another one together and if I plug out the r2 and g2 it has no effect, but if I plug out the D it has.
I have already tried to uncomment the SINGLE_SUB_PANEL line and remake and reinstall the python packages but it has no effect for me. 
I will try to connect the other half panel and work around with the A B C D connections also, maybe just has to swap some of them.

I will write about the results.

hzeller, 2017-07-16:
A little tip: Always first try with the demo programs in `examples-api-use/` before trying Python. You more directly see what is working and what not.

Szili89, 2017-07-18:
I have connected the other half of the screen so now it is 32x64 and like this using the examples-api-use and the python samples they are working properly. But in the future I would like to use just the half panel (16x64). One solution option is to make my content so that it is a 32x64 image and just one half of it contains anything, but it is just a workaround.

Could you please show me the way where should I start to be able to control in python the pixels one by one? I would like to draw some simple graphs on the screen pixel by pixel.

Szili89, 2017-07-19:
I would like to start the programming from scratch but I can not find any manual or datasheet about the SM245TS chip. First I just would like to now what signal should I send to the panel to lit up the led-s. 
Could you please help me how could I start? From where do you find the information to be able to write your code?
Best regards,
Szilárd

hzeller, 2017-07-19:
SM245TS are just something equivalent to 74HCT245. A bus driver.

In order to display pixels, you need to clock them in quickly, switch them on with Output Enable and do that row by row (which you address with A, B, C, C), then go to the beginning. There is no datasheet for the panels, it is just reverse engineered. Look at the code for details.

However, I'd suggest that you first that you play with the library and do some basic programming with it and get comfortable with it do achieve what you'd like to do before attempting to program the matrix directly.

closed 2017-07-19 by hzeller

Szili89, 2017-07-23:
Dear Henner Seller!

I made a clear install and uncomment the SINGLE_SUB_PANEL part and run the make commands. Now it works like it should BUT.
If I restart the pi I always must to run first one of the examples-api-use/ demo and just after that can work the python sample properly
If I run directly the python sample after restart I get the whole image on the upper half of the panel (8x64 layered on the top of each other).
What could cause this problem?

Best regards,
Szilard

hzeller, 2017-07-23:
And I assume you [compiled and installed](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/python#building) the Python interface library _after_ the `SINGLE_SUB_PANEL` configuration ?

In general, this sounds odd. It sounds like the initialization the Python code is doing is not the same as the C++ code is doing, because the 'stuck in upper half' could be due to an address line that is not toggeling. So this could be if the Python library was compiled and installed before the sub panel configuration change.

But then again, if it was that, then even after 'priming' it with one of the demos it should not work better either. Very strange. Can you double-check and go in the `python/` directory and type `make clean` (or `sudo make clean` if there are leftovers written as root) and then `make build-python ; sudo make install` again ?

Szili89, 2017-07-23:
I made your recommendation, but saddly not worked.
So I started again everything. Here is what I made line by line:

sudo git clone https://github.com/hzeller/rpi-rgb-led-matrix.git
sudo nano rpi-rgb-led-matrix/lib/Makefile
_uncomment DEFINES+=-DONLY_SINGLE_SUB_PANEL_
_ctr+x, y, Enter_
cd rpi-rgb-led-matrix/examples-api-use
sudo make
sudo ./demo -D0 -r16 -c2 _(working)_ 

[image]

cd ../python
sudo make build-python
sudo make install-python
cd samples
sudo python rotating-block-generator.py -r16 -c2 _(working)_

[image]

sudo reboot
cd rpi-rgb-led-matrix/python/samples
sudo python rotating-block-generator.py -r16 -c2 _(not working)_

[image]

sudo python rotating-block-generator.py -r32 -c2

[image]

sudo python rotating-block-generator.py -r16 -c2 _(working again)_

[image]

So when I write that I have 32 rows and than going back to 16 it is working.
Do you have any ide what causes this?

Best regards,

hzeller, 2017-07-24:
Thanks for the thorough testing, I think I found what was the culprit. Can you `git pull` again and do a fresh compilation ? It should work better after that.

Szili89, 2017-07-24:
That did the trick! Thanks a lot for the quick solution!

commit f1e6e7d2 (2017-07-23): o Fix initialization issue with panels having only one sub-panel.
  Fixes #356