hzeller/rpi-rgb-led-matrix #357: Ideas on dynamic char spacing with DrawText()

chrisoutwright, 2017-07-17:
I am not sure if the method DrawText() is already using a kind of dynamic spacing (apart from the fixed max width of the font itself) depending on adjacent letters.

Would there be an easier method instead of using DrawText() for every "single character" string with a lookup spacing table like this:

	float letw(char value) {
		switch (value) {
			case 'L': return 2;
			case 'l': return 2;
			case 'r': return 2;
			case 'i': return 2;
		case '|': return 1.4;
		case ':': return 1.8;
		case '\'': return 1.5;
			default: return 2;
		}
	}

And for the main loop, asserting pch is a string for a single row and go its index counter:

int pchlen = strlen(pch);
		for (int go=0; go<pchlen;go++) {
				
				char theval[2] = {0};
				theval[0] = pch[go];

				if (letw(pch[go-1])!=2 && letw(pch[go])==2) 
					{xlan = xlan+(font.width()*(letw(pch[go-1])/2));
				}

				rgb_matrix::DrawText(offscreen_canvas, font, x + xlan, y + font.baseline(), color, theval);
						
				if ((letw(pch[go+1])==2) && (letw(pch[go])==2) && (go != pchlen-1)) {
					xlan = xlan+(font.width()*(letw(pch[go])/2));
				}
				if ((letw(pch[go+1])!=2) && (go != pchlen-1)) {
					xlan = xlan+(font.width()*(letw(pch[go+1])/2));
				}
				
				
			}

It does work in my case but seems like a lot of overhead work,

hzeller, 2017-07-17:
You might want to use the `font.DrawGlyph()` method directly if you are essentially going through the characters character by character (essentially have a look at how [DrawText() is implemented](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/graphics.cc#L28)).

Note that your string-processing is assuming ASCII text, while we are doing UTF-8 here; so you need to use the UTF-8 character iterator to be correct.

Most of the fonts in the fonts directory are fixed width, but for instance the `helvR12.bdf` is a proportional font with varying distance.

chrisoutwright, 2017-07-17:
I've must have overlooked the implementation of DrawText(), when I searched for it with the search function of github. Thank you for the link, this makes it all much clearer. I will try to use DrawGlyph for my case.
Just another question, as pointed out prior. Would you implement a typewriter effect in a similar way like I tried to implement or would you do it altogether differently?

chrisoutwright, 2017-07-18:
Is this what you mean by UTF-8 Iterator?
`char um1[30];
	std::string um2 =u8"ES@K@IST@U@FÜNF";
	strncpy(um1,um2.c_str(),30);
	
    char* whole[] = {um1,"ZEHN@ZWANZIG"}; `

Some special characters like "1 over 4" are displayed but not letters with diaeresis  like ü, also when I use escape characters, it does not work in my case. 
Am I misunderstanding something seriously?

hzeller, 2017-07-19:
You need to properly encode your character as UTF-8. The byte-sequence for codepoint 0xDC is 0xC3 0x9C. Values above 0x7F become a multi-byte sequence ([Read about UTF-8 encoding on wikipedia](https://en.wikipedia.org/wiki/UTF-8)).

It probably works in Python because Python does the right thing on encoding strings internally UTF-8. If you write code with your editor, make sure set your editor to UTF-8 (some really old installations still use Latin1 for instance) or give your string literals the right byte sequence "\u00DC" (if you give u8"..." you still need the text be properly encoded).
(also make sure, if you haven't already, that your locale does UTF-8 by default, e.g. en_US.UTF-8 or de_DE.UTF-8; this is the LC_ALL environment variable).

Don't use wchars, they make life miserable. They are a sad legacy from early Windows implementations...

The UTF-8 iterator is going through a byte string and extract the codepoints [lib/utf8-internal.h](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/utf8-internal.h). It is used internally to the DrawText() function, so I thought you had seen it.

chrisoutwright, 2017-07-19:
Thank you very much for your reply.
At first sight I also thought it could work with \xC3\x9C. Then after testing nearly everything, including testing wide chars, I found my error:

for (int go=0; go<pchlen;go++) {
				
				char theval[2] = {0};
				theval[0] = pch[go];

				if (letw(pch[go-1])!=2 && letw(pch[go])==2 && (go!=0)) 
					{xlan = xlan+(font.width()*(letw(pch[go-1])/2));
				}
				
				rgb_matrix::DrawText(offscreen_canvas, font, x + xlan, y + font.baseline(), color, theval);
				
				
							
				
				
				if ((letw(pch[go+1])==2) && (letw(pch[go])==2) && (go != pchlen-1)) {
					xlan = xlan+(font.width()*(letw(pch[go])/2));
				}
				if ((letw(pch[go+1])!=2) && (go != pchlen-1)) {
					xlan = xlan+(font.width()*(letw(pch[go+1])/2));
				}
		
			}

So reading in every singly character to the DrawText method will naturally destroy any two byte sequence of chars in this way, no matter how they came in!!

I found this when I changed the first row umlaut and the other rows were changed as well, this is sort of segmentation fault maybe. At least I can work at a workaround, maybe use drawglyph. Any ideas how to read in single characters with drawtext or drawglyph which works with umlaut?

hzeller, 2017-07-19:
> Any ideas how to read in single characters with drawtext or drawglyph which works with umlaut?

As I said, use the iterator  I mention above. It goes through your character sequence and extracts the codepoints one after another.

chrisoutwright, 2017-07-19:
Thanks again, I will have a try with the iterator for this matter. Just a single question about the character length. A string like "Olé" will be always counted as length=4 with char type right (not accounting '\0')? 

Could an iterator be modified to express the true length or would I then need the _(in Linux world)_ unloved and unneded wide chars methods like cwslen()?

It would help with the final resolution of the whole matter. Thank you

hzeller, 2017-07-19:
As with regular `strlen()`, which walks each character until you reach `\0`, the equivalent for UTF-8 would be to use the iterator to go through all characters and count them. So this is what you would do if you needed the length of the string (but you don't really in your application as you can just go from left to right). There are also libraries to determine the stringlength of a UTF-8 string (but I encourage you to write the one-liner `while()` loop with the provided `utf8_next_codepoint()` as exercise.

Wide characters are universally unloved not only in the Linux world. They only make sense if they are 32 bit wide and thus only really useful in applications that need direct access to particular characters (i.e. using an index).

The problem with wchar is, that it typically is 16 bit, so it still can't represent all of Unicode (it only worked early 90ies when there were so few chracters defined). So even with that you wouldn't be able to represent a string. You can, by storing a string of uint32_t, but that is somewhat memory inefficient. The beauty of UTF-8 is, that it is compact and still will work typically with old school string operations, as it as well doesn't contain any NUL characters (so e.g. `strcpy()` will just work, or iterating to `\0` if you are only interested in the bytes). Unless you go full 32 bit, don't use wchar's they are a thing of the past and generally painful.

The main problem is, that you attempt to solve your problem by assuming you can index characters, which you can't as UTF-8 is a run-length encoded format. There are two ways to solve this: you either convert everything into a uint32 array with codepoints first (which uses more memory), or you rewrite your code in a way that you only stream through the characters one by one (better solution).

chrisoutwright, 2017-07-19:
I got take a closer look at the method `uint32_t utf8_next_codepoint()`.
As I could infer with the letter Ü (0xc3,0x9c) it combines (c3 fulfilling the first multibyte if condition) it combines it to 0xDC. The iterator `uint32_t cp = *it++;` is in fact already looking at the possible next byte of the multibyte of the utf8_text. 
My single letter method was failing because the second part of Ü (0x9c) was simply not saved in char `theval[2] = {0}; theval[0] = pch[go];`, which I needed to pass along single letters. 
So I would have to make sure the whole multibyte is passed along, and I would be fine.

Thanks for the help so far. This topic can now be closed due to near success. ;-)

chrisoutwright, 2017-07-20:
Just to demonstrate what I described above, the code for the simulated word clock would look like this, using the bad wchar_t fix

Used file to load:

`ES@K@IST@U@FÜNF
ZEHN@ZWANZIG
DREIVIERTEL
NACH@T@VOR@AUS
HALB@G@SECHS@E
ZWEI@S@AM@NULL
H@ELF@FÜNF@VIER
DR@EÜ@N@S@IEBEN
L@ACHT@E@ZWÖLF
ZEH@N@EUN@M@UHR`

`
// This code is public domain
// (but note, that the led-matrix library this depends on is GPL v2)

#include "led-matrix.h"
#include "graphics.h"
#include <iostream>
#include <getopt.h>
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <unistd.h>
#include <time.h>
#include <sys/time.h>
#include <ctime>
#include <stdlib.h>
#include <fstream>
#include <locale>
#include <boost/locale/encoding_utf.hpp>
#include <cwchar>

// These methods are for the bad fix to count umlaute with correct length
using boost::locale::conv::utf_to_utf;

std::wstring utf8_to_wstring(const std::string& str)
{
    return utf_to_utf<wchar_t>(str.c_str(), str.c_str() + str.size());
}

std::string wstring_to_utf8(const std::wstring& str)
{
    return utf_to_utf<char>(str.c_str(), str.c_str() + str.size());
}

using namespace rgb_matrix;

static int usage(const char* progname)
{
    fprintf(stderr, "usage: %s [options]\n", progname);
    fprintf(stderr, "Reads text from stdin and displays it. "
                    "Empty string: clear screen\n");
    fprintf(stderr, "Options:\n");
    rgb_matrix::PrintMatrixFlags(stderr);
    fprintf(stderr,
        "\t-f <font-file>    : Use given font.\n"
        "\t-b <brightness>   : Sets brightness percent. Default: 100.\n"
        "\t-x <x-origin>     : X-Origin of displaying text (Default: 0)\n"
        "\t-y <y-origin>     : Y-Origin of displaying text (Default: 0)\n"
        "\t-C <r,g,b>        : Color. Default 255,255,0\n");
    return 1;
}

static bool parseColor(Color* c, const char* str)
{
    return sscanf(str, "%hhu,%hhu,%hhu", &c->r, &c->g, &c->b) == 3;
[... 457 lines cut ...]
            y += font.height();
			
			
			// This catcher will prevent the code to exit and do other stuff
            if (a == 9) {
                a = -1;
                offscreen_canvas = canvas->SwapOnVSync(offscreen_canvas);
                mb2++;
                sprintf(mb, "%d", mb2);
                rewind(myfile);
                sleep(10);
            }

            a = a + 1;
        }
    }

    // Finished. Shut down the RGB matrix.
    canvas->Clear();
    delete canvas;

    return 0;
}

`

hzeller, 2017-07-21:
You made a very nice clock there!

hzeller, 2017-07-21:
Very nice!

If you are exploring putting a frame around the LED panel, another look I tried is to hide the matrix behind an artist canvas. So it looks entirely white with the typical canvas structure, but then the LEDs shine through:

https://plus.google.com/u/0/+HennerZeller/posts/NgUcGo3YUTz

closed 2017-07-30 by hzeller

bmclain, 2018-03-05:
@chrisoutwright you confirmed my fears that I'll need to manually scroll text myself character by character if I want dynamic coloring. Nice clock!