hzeller/rpi-rgb-led-matrix #358: Combining 2 panels of 32x64

Frank071, 2017-07-20:
Sorry if I am barking up the wrong tree, but am I correct when I say that the library is now properly equipped to drive a panel that does not have 32 columns? I have currently two panels of 32x64 and it seemed I only got this to work after I changed some codes so I could do --led-cols=64. Or did I miss something? If not then I will create a pull request for this.

Further I tried to make a 64x64 display with two of those 32x64 panels which with which I made a simple U chain. I thought the matrix.cfg file would be the core of this, but then I found out that the library ignores this file :-(. Then I found the -L option, but that created a strange pixel layout (image was split in 4 wrongly placed bands of 16 rows ). Checking the code I found some reference to a "parallel parameter" so I tried --led-parallel=2. To my surprise this worked, since that is described as a parameter that defines the number of parallel chains (and I have only 1 chain). Before I try to get into the code, please let me know if I am in error here.

Last, outside scope of subject: is anyone aware of a perl binding for this library? Oh and obviously: great library - apart from these things it was a breeze to get things to work.

Hope to hear

hzeller, 2017-07-20:
The library was always equipped to do 64 pixel long panels from day one. The flag is `--led-chain=2`
If you have a panel that is 64 high and has an 'E' address input line, then you need to wire that up accordingly (see wiring description) and use with --led-rows=64 (but sounds like that is not what you have here). 

The `-U` option is what you want together with --led-chain=4 if you have connected both 32x64 matrix in sequence. The matrix.cfg file is not part of this library - the framebuffer program written by the Adafruit people uses something like that; underlying they are just using this rpi-rgb-led-matrix.

`--led-parallel` works if you have connected the parallel chains like described in the https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md page.

I am not aware of a Perl library, but you are free to do some.

Frank071, 2017-07-20:
Hi Henner, thanks for your prompt reply. I interpreted the value for `--led-chain=2` as "you have two panels chained" and not "the number of times you have 32 rows". This is a great help, I can get rid of my alterations for `--led-cols` then ;-). And yes, that will probably also solve my -U problem (I used -U but forgot to mention and I did it with `--led-chain=2`.

Since I probably need a Perl binding I will look into this and let you know if it results into something.

hzeller, 2017-07-21:
Yeah, the 32x64 essentially behave as two chained 32x32 internally hence the confusion with the chain length.

closed 2017-07-21 by hzeller