hzeller/rpi-rgb-led-matrix #359: Can't make led-image-viewer or video-viewer utilities

tayzerker, 2017-07-22:
1st of all, let me just say that this is amazing.  What you have done is incredible, and we are all really lucky for the hard work you have put in here  - it's an immense help to the entire community.  After trying to get this to work for a week, and trying all kinds of methodologies (adafruit's flavor, a mixture, this version) I settled on trying to do it with your direct master branch, seeming the most up-to-date & powerful.  

By the way, I am using an adafruit hat with 10x  32x32 pixel rgb matrices.  I can get demos running fine, and even alter and run examples.  Anyway, my issue.

I tried to make the image and video viewer of these after installing the recommended libgraphicsmagick and libavcodec.  I found that I didn't have transformer.h or content-streamer.h, and thus make was failing.  So I found them in your github directories and downloaded and copied them into the include directory.  Now the install goes much further, but throws up about 3 pages of errors.  The first one being about expecting a class name before '{'.  But, still nothing useful.  No executable is compiled.

FYI, I have had issues also, when trying to run "image-viewer.py" - claims that there is no module named "rgbmatrix" - perhaps the issues are related?

For context, my goal is to have a series of images fade in and out slowly, cycling through them.  What I have roughly done so far to get to this point (roughly):

sudo apt-get update
sudo apt-get install python-dev python-imaging
git clone https://github.com/adafruit/rpi-rgb-led-matrix
cd rpi-rgb-led-matrix
make
cd utils
sudo apt-get install libgraphicsmagick++-dev libwebp-dev -y
make led-image-viewer
sudo apt-get install libavcodec-dev libavformat-dev libswscale-dev
make video-viewer

maddogs, 2017-07-22:
Try using the current library, not the adafruit library.

git clone https://github.com/hzeller/rpi-rgb-led-matrix.git

tayzerker, 2017-07-22:
@maddogs - Thank you so much for jumping in here.  I'll go back and do that next (unfortunately won't be able to until Monday morning).  I'd been pounding my head on this, but after going back and forth so many times between the adafruit and hzeller repositories, it looks like an adafruit indeed slipped in there, and i totally missed it.  Nice catch!  Thank you!!!  I'll let you know if this solves everything.

tayzerker, 2017-07-25:
So, I did that, and once I got all of the demos and everything running again, I still for some reason am having no luck running image-viewer.py.  Using the exact same syntax as all of the other demos in the python/samples/ directory, I'm getting "ImportError: No module named rgbmatrix"

It does seem to be the only one importing rgb matrix...

Anyway, I navigate into the python/rgbmatrix/ directory and try to run make, but it also complains, "make: cython : Command not found; Makedile:12: recipe for target 'core.cpp' failed"

Any ideas?  Thank you for all of your help!

hzeller, 2017-07-25:
You need to install the python library first before using the python examples.

The steps are described in the [python/ subdirectory](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/python)

You only need to go into python/rgbmatrix/ and have cython installed if you actually want to work on the Python library, so no need do go there.

hzeller, 2017-07-30:
Sounds like you got it working.

closed 2017-07-30 by hzeller