hzeller/rpi-rgb-led-matrix #360: 64x64 display showing every other line

RFburn, 2017-07-24:
I've rewired it a few times now and each time I've had the same problem, but I'm going to try again with pretty much the last combination of changes to try, however, I wanted to ask to get advice on what may be needed to fix my display only showing every other line. Seems like a typical 1/32 scan 64x64 display to me so I'm not sure about the cause, unless it's repeated bad wiring but even wiggling things around while on doesn't even get a brief moment of the other line lighting up.

Images and video can be found at https://photos.app.goo.gl/ROPQQUGKFPN4yHYm2

[image]

Also, are the adapters for sale anywhere, or do you have to get the boards made as the only option currently?

Thanks for everything!

hzeller, 2017-07-24:
So this panel has a strange connection: It has A, B, C, D , but for an 1:32 scan display, one would expect another 'E' line. That isn't there, but instead there is a is an `LA'` which I haven't seen before.
Given that you see every other line, it sounds like it could be that `LA'` is actually A, and A, B, C, D are actually 'B, C, D, E'.

So if you wire it up like that (move `A` over to `LA'`, then shift B, C and D over to A, B, C and add an E connection to D on the panel) and then give it `--led-rows=64 --led-chain=2` (chain = 2 per panel). Maybe this would work ?

RFburn, 2017-07-24:
Would LA not be latch as from the wiring it up guide?
"The strobe signals is sometimes also called latch or lat. We'll call it strobe here."

I'll try what you suggested, but the connector looks exactly how it is described in the wiring guide with the one ground swapped for a E and thats is it.

RFburn, 2017-07-24:
It has
R1 G1
B1 GND
R2 G2
B2 E
A B
C D
CLK LAT
OE GND

hzeller, 2017-07-24:
Yes and it also is at the right position for `Strobe`, so my idea above might not be working.

Ah, you have the pinout. Yes, in this case you need to connect the E line as described in the wiring and then choose `--led-rows=64`

RFburn, 2017-07-24:
That is exactly what I've done and had the issue with every other line missing.  Would that indicate a problem with a particular pin?

hzeller, 2017-07-24:
not sure, the every other pin missing indicates that `A` is not working as intended for some reason. So I'd try to follow that line in particular and see that it is properly connected all the way through (with logic level converter if needed).

RFburn, 2017-07-24:
Will do, thanks for the advice. hopefully it will save some time as rewiring gets time-consuming when remaking adapters. Does every line require level shifting then from 5v to 3.3v? I assume it would be of benefit to shift them all with the raspi being 3.3v and the display being 5v.

hzeller, 2017-07-24:
Yes, I would suggest to always shift the logic level 3.3v -> 5V towards the panel.
(There is an [adapter in this project](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3) that does that).

hzeller, 2017-07-30:
Did this start working ?

RFburn, 2017-08-02:
I did not get it working, I have a bunch of boards arriving tomorrow, active and passive, and fingers crossed that it was a connection issue.

RFburn, 2017-08-03:
Adapters arrived, looks like I may have to try soldering another. New result this time. https://goo.gl/photos/UbdAZ1oZ5E7RXYx47

RFburn, 2017-08-06:
Any tips as to what that sort of display indicates? I'm hoping there is nothing wrong with the boards made from the gerbers, took some explaining to get the files read due to the way KiCad outputs them. Not sure in the end if they did use the design that was shared on PCB Way or the one I submitted after generating new, more compatible, gerbers. Just hoped it indicated something obvious to you so I can try to jumper fix the bodge and track it down before I use yet another board. The first one that missed every other line displayed fine other than that. The adapters are 64x64 compatible, right? Guess I should check the E line.

Thanks again. Got 12 of these boards I'd love to get working with your project rather than the standard receiver card.

hzeller, 2017-08-06:
You are not really giving any information that would help diagnose the problem, e.g. what the parameter are that you are giving to the demo program, nor what your [jumpers chosen for setting the E-line were](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3#choose-e-line-for-64x64-panels-with-132-multiplexing). And of course you need to operate it with `--led-rows=64`.

The image looks like the E-line is not connected properly, as the second half of each half-panel is dark and the parts that are supposed to show in the second half overlay in the first. So the E-line is permanently pulled to ground.

RFburn, 2017-08-06:
Thank you. Sorry, thought there was enough info as I was using your passive adapter currently which is why I had wondered if the current version even supports 64x64 with E-line. For program settings they are all the recommended ones for a 64x64 from the wiki as in my first post --led-rows=64 --led-chain=2 for a single panel demo 0 and timeout 30.

Only thing changed, besides the new display error, is using passive 3 adapter vs hand wiring which has resulted in this change of output. All lines work other than the black bar sections, now I just need to remove those. 

I'll check the adapter boards e-line, if it has one and try again. That's what I was looking to know before repeatedly soldering boards to no avail. Using this until parts for my active boards come. 

Thanks again for the insight. I'm not being oblivious and somewhere on the blog there is an explanation of what behaviour broken connections would have on output, am I? Once I get this working maybe to give back I'll break lines, take photos or have descriptions at least and maybe it will help others focus their troubleshooting efforts. I guess I'd still have to explain it for other board layouts, so maybe not so universally helpful as I first thought. Would be nice if something like that existed or at least described issues that arise from particular lines being disconnected. That's a lot of work though and I'm sure you're busy. Once I get these first ones working maybe I'll grab other scan models and write something up, if that's at all useful or helpful. I guess visually it would be daunting to show each problem, especially if multiple exist, but maybe a table would be possible. Anyways, I'll try that after and if it seems or any use I'll pass it along and you can do with it what you will. Just want to say that thanks. 

I'll go mess with the passive boards e-line now.

hzeller, 2017-08-06:
If you use the active board, it has the E-line, but you need to configure it with jumpers to output on Pin 4 or Pin 8. In your display it is Pin 8

https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3#choose-e-line-for-64x64-panels-with-132-multiplexing

hzeller, 2017-08-06:
The passive adapter does not support the E-line. I would not suggest to use the passive adapter, as you want to level shift the signal to the 5V needed by the panel (as they are not guaranteed to work with the 3.3V of the Raspberry Pi), and you also don't want to stress your Pi outputs to drive long cables,

RFburn, 2017-08-07:
Well, then that explains why the e-line isn't connected. I was only using the passive for some temporary testing while awaiting parts for the active board. Cables are pretty short but I guess I'll wait until the parts arrive. 

Wish I had realized the passive board didn't support the e line, but I probably missed where it said so. Oh well, It was only $5.  Fingers crossed for smooth sailing after all this fail.

Thanks again, I'll close this and stop bothering you.

closed 2017-08-07 by RFburn

RFburn, 2017-08-08:
Ok, I lied, I'll bother you one more time just in case someone else decides to search this issue after not realizing the Passive adapter is not up to date for 64x64 panels with the E line and mistakenly orders a bunch only to find they don't work. Modifying the Passive adapter has resulted in the displays working flawlessly in a string of four with 3 rows. I know, level shifting, pi stress, bad idea, etc but I take full responsibility and have Pis to spare.

I Had to cut the traces going to the 6th pin on each connector, top and bottom, then added a jumper across pins 4 & 16 to repair the broken grounds and jumpered pin 6 on each connector to pin 10. I have black boards so that made the traces hard to see but using the OSHPark image of the board layout made it easy to see where exactly to cut. I mistakenly did one extra jumper but I did the two ground jumpers and just did the extra one as I must not of check continuity on that one. No harm with it anyways. 

Here is a image if anyone needs it:
[image]

Maybe a message on the passive adapter files would be useful so others don't suffer the same fate. I'd update the files and share them but I don't know KiCad. My bad for assuming the board supported it since the software supported it.

hzeller, 2017-08-09:
Alright, updated the readme
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/adapter/passive-3/README.md

sachasayan, 2020-08-02:
Just wanted to say nice work here, @RFburn, as I'm trying to similarly adapt the passive boards. Thanks for leaving a trail of knowledge here. 

One clarification: Your written directions say you were cutting the traces on pin 6. 

By your own diagram, and by hzeller's documentation [on the wiring page](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md), I believe that should actually be pin 8 (or, depending on the board, pin 4), correct? 

[image]