hzeller/rpi-rgb-led-matrix #362: Gerber format

RFburn, 2017-07-26:
I am having an issue with submitting the gerbers to PCBWay for production. They attempt to open the files only to get errors and nothing displays. The format they are expecting is RS-274X format, does that match with the format of the files. I seem to see it does. Looking further I found it seems to be an issue with CAM350 and KiCad produced gerber files. They require the "extended attributes" option is off and G04 #@! removed from the start of the file oddly enough. So I am wondering then if the files were produced with extended attributes off? I don't use KiCad and am unsure my way around it but would love to get active and passive boards made up.

hzeller, 2017-07-26:
So I suggest either you load it into KiCAD and generate whatever they need or you could ask them if they can take the KiCAD file.
(I don't remember the settings I used for outputting it)

maddogs, 2017-07-26:
I had the same problem with PCB way. The passive board was fine, the active came back with an error. However after a few messages back and forth, they did make my order with out any issues.

If it helps, I've shared my order here

https://www.pcbway.com/project/shareproject/W13095AS3_plot.html

RFburn, 2017-07-26:
Thanks for the maddogs, sadly for some reason it gets rejected right away. So weird given you've produced it before. I'm going to talk to them a bit else I'll just try another fab house.

Sent the message they couldn't open the file, which seems very weird since you've had it made. Hopefully I can get it sorted out. Always good to try somewhere new.

It is due to a bug with KiCad and CAM350 failing to read the format. 

Thanks again both of you.

beikimajid, 2017-07-27:
Hi
I have the same problem.I use "Altium designer" and import gerber file but I can not convert gerber to pcbdoc.Is it possible to attach the ".PCBdoc" ?

hzeller, 2017-07-30:
pcbdoc seems to be a proprietary Altium format.

The PCB is designed in KiCAD which you should just fire up and load the `active3-rpi-hub75-adapter.kicad_pcb` in `pcbnew`, then export into whatever format needed. All the source files for the PCBs are in the [adapter directory](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3).

closed 2017-07-30 by hzeller