hzeller/rpi-rgb-led-matrix #364: image viewer sample as a function

arcadeperfect, 2017-07-29:
Hi,

I can't get your image viewer sample code to work when wrapped in a function. This works:

`img = 'drink.png'
image = Image.open(img)
options = RGBMatrixOptions()
options.brightness = 40
options.rows = 16
options.chain_length = 1
options.parallel = 1
options.hardware_mapping = 'regular'
matrix = RGBMatrix(options = options)
matrix.SetImage(image.convert('RGB'))`

But this doesn't:

`def loadImage():
	img = 'drink.png'
	image = Image.open(img)
	options = RGBMatrixOptions()
	options.brightness = 40
	options.rows = 16
	options.chain_length = 1
	options.parallel = 1
	options.hardware_mapping = 'regular'
	matrix = RGBMatrix(options = options)
	matrix.SetImage(image.convert('RGB'))

loadImage()
`
Is there something about the way functions execute that prevents this from working?

Thanks

hzeller, 2017-07-29:
Well, after the function returns, your RGBMatrix is gone.

So you need to make sure to have a RGBMatrix instance that is there for the runtime of your program (so somewhere in main), and then have the loadImage() function get pointer to that instance.

closed 2017-07-29 by hzeller

arcadeperfect, 2017-07-30:
hey, thanks for the reply. making progress...