hzeller/rpi-rgb-led-matrix #365: No signal on GPIO pins

dsanderson, 2017-07-30:
Hello, 

Thanks for making this library! We're trying to use it to drive several 32x64 matrices, but they aren't lighting up at all. We've tested the matrices with an arduino mega, and they work fine, so we suspect there something wrong with our adapter board (the active 3) or how we're using the library.  I manually checked all the GPIO pins on two different pis running demo -D 0, but didn't see a signal on any of the pins going to the panels. I'm new to GPIO on the raspberry pi, so do you have any advice on what may be wrong? Should I even expect to see a signal? Is there a global setting I'm missing somewhere? Thanks!

hzeller, 2017-07-30:
Make sure to read the [troubleshooting section](https://github.com/hzeller/rpi-rgb-led-matrix#troubleshooting). In particular a minimal installation with no components that also try to access the GPIO pins (no sound enabled, no 1 wire connection).
What Raspberry Pi are you using ?
You should see a signal on the GPIO pins, if you have an image to show, they will constantly change, so you should see it on the oscilloscope (not on a multimeter though, they are too fast).

So you should narrow down where things happen: does the GPIO pin not toggle, or is the value not coming through the level shifters ?

dsanderson, 2017-07-30:
Hi,

The GPIO pin does not seem to toggle; I'm checking with my oscilloscope on the rpi pins.  I've tested on an RPi 3 with the sound and all interfaces turned off, running Raspbian, and I've also tested on an RPi 2 running Raspbian Lite (The sound module was loaded though, I ran with --led-no-hardware-pulse).  I also can't seem to toggle the pins through the RPi.GPIO python library, so I assume the problem is outside the led matrix library, but I'd appreciate any advice you have on debugging what's causing this.  Thanks.

dsanderson, 2017-07-30:
Haha, I feel like an idiot, my oscilloscope was broken.  The RPi's are fine, there were some bad solders on the adapter board.  Thanks for trying to help, and sorry for the mistake on my side.

closed 2017-07-30 by dsanderson