hzeller/rpi-rgb-led-matrix #366: Compile error

Edragon, 2017-07-31:
Hi!

I am not sure if it is an error, 

make
g++ -I../include -Wall -O3 -g -fPIC  -DDEFAULT_HARDWARE='"regular"' -c -o bdf-font.o bdf-font.cc
bdf-font.cc: In member function ‘bool rgb_matrix::Font::LoadFont(const char*)’:
bdf-font.cc:88:37: error: expected ‘)’ before ‘PRIx64’
bdf-font.cc:88:72: warning: spurious trailing ‘%’ in format [-Wformat]
bdf-font.cc:88:72: warning: too many arguments for format [-Wformat-extra-args]
Makefile:146: recipe for target 'bdf-font.o' failed
make: *** [bdf-font.o] Error 1

line 87 in bdf-font.cc:

    else if (current_glyph && row >= 0 && row < current_glyph->height
             && (sscanf(buffer, "%" PRIx64, &current_glyph->bitmap[row]) == 1)) {
      current_glyph->bitmap[row] <<= bitmap_shift;

"%" PRIx64 seems should be "%x", compare to your other version. I fix it and compile well. If it is not an issue, please just close it. Thank you!

hzeller, 2017-07-31:
What

Edragon, 2017-08-01:
Hi, use not the latest version, if it is the distribution problem, I will download a fresh new raspbian pi to try again.
Use "Git clone" this project should be correct? do I need add "recursive" or submodules? 

The only problem I get now, is the 1/8 scan rate can not work, the output show 4* 1/4 in the panel, and each 1/4 is still not correct, seems rotated can not really see the problem .. I will try other more panels in next day.

Thank you.

hzeller, 2017-08-01:
Can you try again with your current distribution ? I changed something that should work even with older compilers.

Sounds like you got a snake multiplexing; other people already wrote a transformer for this, have a look at the bugtracker where people posted their solutions.

Edragon, 2017-08-01:
Hi, still not working, probably need the latest distribution:

root@work:/home/pi/rpi-rgb-led-matrix# make -C example*
make: Entering directory '/home/pi/rpi-rgb-led-matrix/examples-api-use'
g++ -I../include -Wall -O3 -g -c -o demo-main.o demo-main.cc
make -C ../lib
make[1]: Entering directory '/home/pi/rpi-rgb-led-matrix/lib'
g++ -I../include -Wall -O3 -g -fPIC  -DDEFAULT_HARDWARE='"regular"' -c -o bdf-font.o bdf-font.cc
In file included from /usr/include/c++/4.6/cinttypes:35:0,
                 from bdf-font.cc:22:
/usr/include/c++/4.6/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the upcoming ISO C++ standard, C++0x. This support is currently experimental, and must be enabled with the -std=c++0x or -std=gnu++0x compiler options.
bdf-font.cc: In member function ‘bool rgb_matrix::Font::LoadFont(const char*)’:
bdf-font.cc:89:37: error: expected ‘)’ before ‘PRIx64’
bdf-font.cc:89:72: warning: spurious trailing ‘%’ in format [-Wformat]
bdf-font.cc:89:72: warning: too many arguments for format [-Wformat-extra-args]
Makefile:146: recipe for target 'bdf-font.o' failed
make[1]: *** [bdf-font.o] Error 1
make[1]: Leaving directory '/home/pi/rpi-rgb-led-matrix/lib'
Makefile:18: recipe for target '../lib/librgbmatrix.a' failed
make: *** [../lib/librgbmatrix.a] Error 2
make: Leaving directory '/home/pi/rpi-rgb-led-matrix/examples-api-use'

hzeller, 2017-08-01:
what is your distribution ?  ` cat /etc/os-release`
What is the output of `g++ -v` ?

hzeller, 2017-08-01:
Don't update your compiler quite yet, can you have another attempt with the latest version ?

Edragon, 2017-08-01:
PRETTY_NAME="Raspbian GNU/Linux 7 (wheezy)"
NAME="Raspbian GNU/Linux"
VERSION_ID="7"
VERSION="7 (wheezy)"
ID=raspbian
ID_LIKE=debian
ANSI_COLOR="1;31"
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

and

Using built-in specs.
COLLECT_GCC=g++
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/4.6/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../src/configure -v --with-pkgversion='Debian 4.6.3-14+rpi1' --with-bugurl=file:///usr/share/doc/gcc-4.6/README.Bugs --enable-languages=c,c++,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.6 --enable-shared --enable-linker-build-id --with-system-zlib --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.6 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --enable-plugin --enable-objc-gc --disable-sjlj-exceptions --with-arch=armv6 --with-fpu=vfp --with-float=hard --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
Thread model: posix
gcc version 4.6.3 (Debian 4.6.3-14+rpi1)

Edragon, 2017-08-01:
Hi,
The problem fixed with your last code update! thanks.

commit 35b0648f (2017-08-01): o Even more bending over backwards for old compiler installations
  to work. Fixes #366  (maybe now).
closed 2017-08-01 by Edragon