hzeller/rpi-rgb-led-matrix #367: Question about 64*64 matrix

Einhornkot, 2017-08-01:
Hi,

I think about to build a led matrix with my raspberry and an Adafruit Hat, but there are some questions that I want to get solved before I'm going to order the parts:

Is it possible to connect two 64*32 panels in a chain to create a 64*64 panel (like in the picture below) which can get connected to the Adafruit Hat?

I'd be happy about any sort of help.

Philipp :D

Picture:
https://ibin.co/3VPz7CBBEZ9D.png

hzeller, 2017-08-02:
Yes.
Please make a minimal effort reading the documentation first before sending a message that 70+ people are receiving.

Each panel internally is a chain of two, so you use `--led-chain=2 --led-rows=32`; your final chain will be `--led-chain=4`. Then you want to [remap the pixels](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use#remapping-coordinates) that they are nice and upright, and you can use the builtin UArrangementTransformer for that.

MaitreDede, 2017-09-16:
Hi @hzeller,

In my case, I have a single [64x64 matrix](https://www.aliexpress.com/item/Coreman-HD-full-color-p2-5-160x160mm-64X64-smd-led-display-big-module-p10-p12-p16/32664271760.html?spm=a2g0s.9042311.0.0.yhFIkZ). 
I don't know which arguments I have to configure to properly run the demos.

For wiring, I have plugged the panel's connector pins to gpio using "regular" pinout (pi 3 gpio).
Then I launched demo 0 (rotating square) with `sudo ./demo --led-no-hardware-pulse --led-show-refresh --led-rows=64 --led-gpio-mapping=regular-pi1 --led-slowdown-gpio=2 -D 0 --led-chain=2`.
You can see resulting video here : https://youtu.be/lv6C89j0i1M

What can be the problem, and how may I fix it ? Thanks.

closed 2018-01-21 by hzeller