hzeller/rpi-rgb-led-matrix #368: Vertical font rendering (rotated by +/- 90 degrees)

mb21, 2017-08-04:
I saw the `VerticalDrawText` function, but it prints each character non-rotated.

    A
    B
    C

What I'm looking for is to have the line growing down as well, but each character rotated. Looks like we'd have to modify the [Font::DrawGlyph](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/bdf-font.cc#L162) function or create a `Font::DrawRotatedGlyph` function.

Is there a better way? Or would you consider implementing this or merging a pull request?

hzeller, 2017-08-04:
This is not good to put in the DrawGlyph. Because then you have 90 degrees, what if someone wants 45 degrees ?
The better solution for this is to have a delegating canvas that does rotation for you. You can even re-use the transformers
```
RGBMatrix *matrix = ...
RotateTransformer rotater(90);
rgb_matrix::DrawText(rotater.Transform(matrix), ....)
```

mb21, 2017-08-04:
Thanks, this looks promising! I assume this is not available in the python API, so I'll give it a try and switch to C++

closed 2018-01-21 by hzeller