hzeller/rpi-rgb-led-matrix #369: How to center a video using SetImage - is it possible?

BNNorman, 2017-08-05:
How can I change the top row to position and image at any x,y position?

I modified your sample Image-viewer.py by using opencv to read in an MP4 video and converted each frame to PIL format and used SetImage to output it to screen - it works rather well. I can get 25 fps (probably more as I had it running at double speed at one point) using a Pi 3 and your Active Adapter. However, the video frames start at top left and being a wide screen recording I'd quite like to position it lower, in the middle of the display. I'm using 2 of 32x64 1/16 scan P5 panels arranged as a 64x64 matrix - just for personal curiosity - something to while away my retirement.

Here's my code (running on Python 2.7) if anyone else wants to have a go.

And thanks again for all your work.

# show a video by streaming the frames to the matrix
from PIL import Image
import time
import cv2
import numpy as np
from matplotlib import cm

from rgbmatrix import RGBMatrix, RGBMatrixOptions

options=RGBMatrixOptions()

# the matrix is two 32x64 panels in parallel
options.rows=32
options.chain_length=2
options.parallel=2

matrix=RGBMatrix(options=options)

source="./Xtremis.mp4"  # a little bit of Dr Who

# make sure we can read the source
try:
	fp=open(source)
except:
	print"Unable to open ",source
	exit(0)
fp.close()

vid=cv2.VideoCapture(source)

FPS=25 # this can be obtained from the actual video but I'm being lazy here
FRAMETIME=1.0/FPS

while True:
    t0=time.time() # frame start
    
    # grab the next frame
    (grabbed, frame) = vid.read()
    if not grabbed:
        print ("Frame not grabbed (maybe end of video)!")
        break

    # convert from numpy to to PIL image
    im=Image.fromarray(frame) 
    im.thumbnail((matrix.width,matrix.height),Image.ANTIALIAS)
    
    # display it
    matrix.SetImage(im.convert('RGB'))
    
    # wait for next frame refresh
    while(time.time()-t0)<FRAMETIME:
        pass

input("Press any key to terminate")

hzeller, 2017-08-05:
SetImage() actually has to additional optional parameters, `offset_x` and `offset_y`

```
matrix.SetImage(im.convert('RGB'), 12, 5)
```
places the image at position (x,y) = (12,5)

BNNorman, 2017-08-05:
Ah, brilliant - thanks for such a quick response - works a treat.
Great work!

closed 2017-08-05 by hzeller