hzeller/rpi-rgb-led-matrix #371: File permission denied after matrix=RGBMatrix() call

BNNorman, 2017-08-07:
I thought I'd modify the python/samples/image-viewer.py script to cycle through a list of images and display them in a round robin fashion. I've called it slideshow.py, of course. However, I get permission denied when I run the script using sudo python slideshow.py at the point where it attempts to open the image files but ONLY within the code that follows the matrix=RGBMatrix(options=options). If I comment out that statement and the matrix.SetImage() statement the loop runs just fine with no errors.
I'm running it on a Pi 3. All the images have permissions 777.

Your original code loads the image before the matrix = RGBMatrix(options = options) statement. I did try moving the display code within the for loop into a method (def showImage) which was defined before the matrix instantiation but I still got a permission denied error. 

I then placed os.getegid() calls before and after the matrix= statement and lo-and-behold it changed from 0 to 1. So it looks like you RGBMatrix call is clobbering the effective user.

I will try a workaround of loading the images into a list before the loop but that's going to be memory hungry with a lot of images.

Here's my code if you want to try it.

#!/usr/bin/env python
'''
modified image-viewer.py
cycle through a list of images found in config.txt
and display on rgb matrix
'''
import time
import sys
from rgbmatrix import RGBMatrix, RGBMatrixOptions
from PIL import Image

# open the list of images to display
try:
	images=open("config.txt")
except:
	print "Unable to open config.txt"
	sys.exit(0)

# Configuration for the RGB LED matrix
options = RGBMatrixOptions()
options.rows = 32
options.chain_length = 2
options.parallel = 2
options.hardware_mapping = 'regular' 
options.gpio_slowdown=2

print "egid before=",os.getegid()

# after this line I get permission denied trying to open the images
matrix = RGBMatrix(options = options)

print "egid after=",os.getegid()

for image_file in images:
	image_file=image_file.strip() # remove trailing cr/lf

	# this code fails with permission denied 
	# if the matrix = RGBMatrix(options = options) is commented out
	# the code opens the image ok.
	
	try:
		image=Image.open(image_file)
	except Exception as e:
		print "Unable to open ["+image_file+"]",e.args
		continue

	# Make image fit our screen.
	image.thumbnail((matrix.width, matrix.height), Image.ANTIALIAS)
	matrix.SetImage(image.convert('RGB'))

	try:
    		print("Press CTRL-C to stop.")
        	time.sleep(5)
	except KeyboardInterrupt:
    		sys.exit(0)

print "Show finished"
sys.exit(0)

hzeller, 2017-08-07:
The reason is, that the library drops the privileges from root to daemon, once it has initialized the hardware. This adds some  safety in case something goes wrong. But it also means, that other interactions (e.g. reading images) are then done as a different user.

You can switch that off with `--led-no-drop-privs` or the equivalent option when starting the matrix.
You also could just simply make your images readable to everyone `chmod a+r <images>`.

BNNorman, 2017-08-07:
Thank you again for your quick replies.

closed 2018-01-21 by hzeller

digital-cinema-arts, 2019-08-16:
I'm having the same problem but with calling from the python bindings.  Is there a switch for 'led-no-drop-privs' in the python interface?  I haven't found it.

digital-cinema-arts, 2019-08-16:
I've posted the issue on Adafruit's forum:

https://forums.adafruit.com/viewtopic.php?f=47&t=155192

digital-cinema-arts, 2019-08-16:
Found it after grepping core.pyx:

 RGBMatrixOptions.drop_privileges = False