hzeller/rpi-rgb-led-matrix #372: flaschen-taschen

beikimajid, 2017-08-13:
Hi
flaschen-taschen server does not have -L Parameter.
I wrote U-arrangement(Transformer) and I used on led-matrix.cc .
but when use ./send-text from client, It does not work and it only appears on the first module.
what's the solution to use U-arrangement Or S-arrangement?

hzeller, 2017-08-13:
You are not describing what you are doing (e.g. how you added the UArrangementTransformer) nor what options you passed to the `ftserver` when starting nor anything else that could be useful in figuring out why you are only getting a limited output.

Also, did you use the `-g` option on the `send-text` client to set the display size ? By default it is 45 wide, which is probably smaller than your display.

Please include the relevant information.

hzeller, 2017-08-13:
(why did you file this under rpi-rgb-led-matrix and not https://github.com/hzeller/flaschen-taschen ?)

beikimajid, 2017-08-14:
I do not know where the problem is.
From rpi-rgb-led-matrix or flaschen-taschen.
I use P5 module(32x64 1/16 scan) in 3 Rows & 6 Columns.(384px X 96px).

I wrote U-arrangement(Transformer) and I used on led-matrix.cc( Applytransformer)

My commands for server are:
 make FT_BACKEND=rgb-matrix
./ft-server

My commands for client are:
export FT_DISPLAY=localhost
./send-image some-image.png 

How can i do scale my image to full screen?

hzeller, 2017-08-14:
Again, hyou need to give the `-g` parameter to the client program.

closed 2017-08-15 by hzeller