hzeller/rpi-rgb-led-matrix #373: led-image-viewer for large gif

beikimajid, 2017-08-14:
Hi
I use led-image-viewer from utils.
When I send gif image(more than 6MB) I get std::alloc error.This is probably an error for RAM.How can I solve it?

hzeller, 2017-08-14:
If this is a gif file with many frames, then this might happen as all frames are preprocessed in memory to allow for fast access, and the Pi has only limited amount of memory.

Read the [documentation of the `led-image-viewer`](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/utils#image-viewer): You can use the `-O` option to create a preprocessed stream onto disk, which you then can read from there; that allows to use arbitrary long.

beikimajid, 2017-08-14:
Hi
I use -O but for 6MB GIFs, get 1G disk space and will be slow when displayed.
Is there a way to compress this?