hzeller/rpi-rgb-led-matrix #374: Active-RPi-3xHub75-Adapter This is hopefully a dumb question....

fusionfoto, 2017-08-14:
... So I wanted to make life easy on myself and try out this led matrix stuff as a fun project.

So I ordered the Active RPi Adapter boards from Oshpark and they look great.

However, they don't look like this board:
[image]

They look like this:
[image]

So I'm a little confused. Am I supposed to be acquiring and soldering on three ICs to these boards? I could have just breadboarded them if that's the case....  I definitely have to add headers in any case. 

These are multilayer PCBs, so its not obvious to me -- I haven't looked at the schematic to this level of detail.

thanks in advance!

NetForces, 2017-08-14:
OSHPark is a PCB supplier not a complete solution supplier. So yes you will receive only the PCB (purple part) and will need to buy the buffers, connectors and passive (capacitors and resistor) and solder them.

fusionfoto, 2017-08-14:
ok. thanks for the quick response!

closed 2017-08-15 by hzeller