hzeller/rpi-rgb-led-matrix #376: 32x32 panel strange pixel mapping

vinchi9, 2017-08-17:
Hello,
I am using [this](https://www.digikey.cz/product-detail/en/mikroelektronika/MIKROE-2347/1471-1626-ND/6137614) 32x32 panel. After first connection the panel displayed everything very strangely. I tried multiple parameter configurations and I figured out, that the scan mode is 1:8, but the pixel mapping is odd. The panel is divided to 4 segments and some are badly placed and rotated. This is the input image: 
[image]
I used this command:
`/led-image-viewer --led-rows=16  --led-chain=2 /home/pi/a.png -L`
And this is the result:
[image]

How can I fix this?
Thank you

vinchi9, 2017-08-18:
In the end I solved the problem by creating a remapping filter in ffmpeg and converting my media files. Thank you for creating this library.

closed 2017-08-18 by vinchi9

hzeller, 2017-08-19:
Cool. Yes, this is as well a way; the Transformer is essentially a remapping filter as well.