hzeller/rpi-rgb-led-matrix #377: Wiring diagram

scandinaro, 2017-08-18:
I’m having trouble understanding the wiring diagram with the emoji etc. Is there a more simple version? I’m connecting a pi to the 16x32 board directly.

hzeller, 2017-08-18:
You are only connecting a single chain, so just follow the :smile: symbol and connect all the wires to the corresponding names.
(The additional emojis are for the second and third chain)

scandinaro, 2017-08-18:
Thanks @hzeller I understood that much (or thought I did).  We have the 16x32 rgb led matrix connected to the raspberry pi what we thought was correctly based on following the 😄  emoji only, but the syntax about what wasn't required for single chain was confusing.  We skipped the strobe as we didn't see that mapped on the matrix, but were not sure if that was correct.  When we try to run the demos we don't get any error, but nothing is output to the matrix so I'm assuming we have something wired incorrectly.  We had originally followed the adafruit wiring directions which didn't work, then tried your directions in the readme and had no further luck.  Our matrix also doesn't have the input pins marked so we were assuming they matched the mappings we could find on the internet.  Any additional guidance you can provide would be extremely helpful.

hzeller, 2017-08-18:
Yeah first off: don't use the Adafruit wiring, they have a very old tutorial that describes an ancient version of my library. Use this current library with its wiring description.

You can't skip strobe, otherwise nothing will happen. You need to connect all of the :smile: pins except `D`, as you only have a 16 high matrix that requires addresses `A`, `B` and `C`. If your connector does not have any description then the image in the wiring.md is your best guess.

A matrix typically has two connectors, one input and one output. Make sure to connect to the right one.

If it is a simple matrix, the relevant flag to the demo program would be `--led-rows=16`.

Note, directly wiring will only work if your matrix happens to work with 3.3V digital signals. For many, you need to have 5V digital signals, so you need a level converter. See the main README.

closed 2018-01-21 by hzeller