hzeller/rpi-rgb-led-matrix #379: Rpi 2 Sound problems

malublu, 2017-08-19:
Hello,

i set dtparam=audio=off in /boot/config.txt
and i plugin an external usb sound card.
But if i play any animation with the library (demo) than i hear some creapy sounds.
How can i fix it?

hzeller, 2017-08-19:
Don't connect a speaker :)

The issue is, that the hardware that is connected to the audio output needs to be re-purposed for exact timing generation in the LED matrix library (this is why you need to disable audio, so that the led matrix can use it). Since the audio connector is still connected to that part of the hardware, you hear the noises.

Since you can't use the audio anyway, just disconnecting a speaker is the best way to deal with it.

If you need audio, you can still just add a USB sound-adapter (and that audio quality is much better anyway). You an also switch off using the hardware pulse generation in the library (with the `--led-no-hardware-pulse` option), but then you loose the advantages.

closed 2017-08-19 by hzeller

malublu, 2017-08-20:
Jear i already using an USB sound-adapter but i still hear the noises.
[image]
with dtparam=audio=off in /boot/config.txt

I only hear the noieses if i run the code.
So whats now 😜 

I also try: --led-no-hardware-pulse. But i  i still hear the noises.

Is there any configuration that i not seen?

hzeller, 2017-08-20:
In that case I assume that your power supply of the Raspberry Pi is not creating a stable 5V. If you have connected the Matrix to the same 5V as the Raspberry Pi without much filtering in-between, then this could happen as the Matrix stresses the power quite a bit and can make it swing with glitches between 4V and 5V in high frequency. This kind of noise the Pi will be ok with, but you'll hear that in the analog output, as the 5V go directly to the USB converter.

Measure the 5V input on the Raspberry Pi with an Oscilloscope to see.

Another way to test: You can try to operate them on different supplies; don't use the HAT to distribute the same power, just connect the Matrix to your high power power supply and only connect Ground with the Matrix and HAT. And then the Raspberry Pi you just power with a regular 5V USB supply, so that both circuits use a different 5V supply. Important is to have Ground (-) connected to both.

Also, depending on how much USB bandwidth your audio output needs, it might compete on the bus for time to send the data, though that is probably less likely

hzeller, 2017-08-20:
I see that you have the USB adapter connected via a cable. So you can also stabilize power there, by putting an inductor in series of the 5V line to the USB adapter and having a couple of 1uF ceramic capacitors and a large (1000uF or more) low ESR capacitor in parallel from 5V to ground.

Or cut the 5V going to the USB adapter and provide your own.

In particular it looks like you are operating the speakers directly from the USB, so you will definitely hear noises on the power supply line.

malublu, 2017-08-23:
@hzeller near you're right.
I test it with a second external power supply (USB).
And now all is fine.

Thanks ;)

Now I cut an USB extension cable and solder it.

_Gesendet von meinem OnePlus ONEPLUS A5000 mit [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_