hzeller/rpi-rgb-led-matrix #380: strange color issue with top and bottom half of panels

digital-synapse, 2017-08-20:
Not sure what i'm doing wrong. I have an 64x32 panel which is really 2 32x32 panels chained internally i think. However i have this strange color issue on the top and bottom half of the panel. ANy ideas?

https://youtu.be/7ghCKmX0qp4

hzeller, 2017-08-20:
You have not described how you connected them ? Did you manually wire them up or used a level converter ? Is it a Pi 1 or Pi 2/3 that you have connected it to ?

digital-synapse, 2017-08-20:
Sorry. This was wired manually directly on a pi zero. I also connected all
of the grounds

hzeller, 2017-08-20:
So directly wiring only works well if the LED panel in question can deal with 3.3V digital inputs, not all of them can and want 5V logic levels; and it could well be that some of the inputs are better than others to digest the voltage. So you might need to consider adding a level shifter.

And of course double checking every wire, but I suspect you already did this about 3 times :)

digital-synapse, 2017-08-20:
yes lol rechecked all my wires 3+ times. disconnected everything and started over with another pi zero. everything works as expected now. looks like i have some bad GPIO pins on by prototype board. Does anyone know if there a c# wrapper for the api?

closed 2017-08-20 by digital-synapse

hzeller, 2017-08-20:
I am not aware of a C# API, but if you want to create a language binding, you might want to wrap the C-API, as this usually is more easy than wrapping a C++ API with another language.

https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/include/led-matrix-c.h