hzeller/rpi-rgb-led-matrix #387: Passive-3 adapter and U-arrangement

beikimajid, 2017-08-29:
Hi
I use Passive-3 adapter. My wiring is : 

P2 < P1 
         ||
P3 > P4 -------- parallel1

P6 < P5
          ||
P7 > P8---------parallel2

P10 < P9
            ||
P11 > P12------parallel3

I want use [./led-image-viewer] and And show the image with [ --led-parallel=3] and [-U] parameter.
How can I do that?

hzeller, 2017-08-29:
Try the option `-L` (large display), but first give the options for the length of the chain. The chain length is 8, as you have four displays with 64 width each:

  `--led-chain=8 --led-parallel=3 -L`

beikimajid, 2017-09-02:
I have two U-arrangement and one S-arrangement (Asymmetric) ,as example
P2 < P1
||
P3 > P4 -------- parallel1

P6 < P5
||
P7 > P8---------parallel2

P10 < P9
||
P11 > P12
            ||
P14 < P13------parallel3

Can the image be continuously displayed in all modules?

hzeller, 2017-09-02:
Yes. You essentially need to consider all chain to be 6 panels long (but in two of them you only do two).

Then you write a Transformer that maps your panels to whatever final layout you choose.

closed 2018-02-17 by hzeller