hzeller/rpi-rgb-led-matrix #389: 64x32 panel not working

szwenni, 2017-09-01:
Hi

First: Great work !!

But my panel isn't working. I connected everything like you described at the wiring diagram, but the panel show nothing. Not even some flickering or anything.
As far as I've seen this the panel is using hc475 chips. They are using 5v logic level (?). So is this possible that I don't even see a flicker if logic level is wrong ?
I think that the process is running on 100% cpu is normal.

Best Regards,
Sven

Kartheek1512, 2017-09-02:
I am in the same boat. let me know if you figure out.

hzeller, 2017-09-02:
Yes, if they are using HC chips, not HCT, it might very well be that they don't react to the 3.3V level output by the Raspberry Pi and it might well be that you don't see anything; so in that case I suggest to solder up an [active adapter](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3).

(If you have a little older checkout of this library but a recent Debian Stretch Raspbian distribution it could as well be that you don't see anything (and other wierd things happen, such as you can't Ctrl-C the process). In that case make sure to do a fresh `git pull` to get the latest.)

szwenni, 2017-09-03:
Hi,
I updated the bib and it worked. Thank you.

One other question. Is there any documentation how these panel are working ?
I am trying to understand your code, but some general explanation on how is one panel working and what do you need to to do to control two daisy chained ?

Br,
Sven Kröger

Kartheek1512, 2017-09-04:
i am connecting it to raspberry pi 3. I connected its gpio pins to the LED
directly as given in the instructions. I dont see any LEDs light up. Do you
think I have to use this hat below to convert to TTL logic.

https://www.adafruit.com/product/2345

Thanks,
Kartheek

hzeller, 2017-09-04:
Which instructions did you use ? I am asking, because some people accidentally use the instructions from the Adafruit tutorials (which are outdated and wrong) instead of the ones in the project directly:

https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md

You can use the Adafruit HAT, but it is only useful if you only connect one chain (then run with `--led-gpio-mapping=adafruit-hat`). Better is to solder up this adapter for up to three chains provided in [the rgb led matrix project itself](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3#adapter-pcb-to-support-up-to-3-panel-chains).

hzeller, 2017-09-04:
@szwenni I think there is no real description anywhere. But in general, the pixels are clocked in in two rows (hence RGB 1 and 2) as a shift register, then they are latched in and switched on for a while. Then the next pair of rows etc (the row-pair in question is addressed with the ABCD address line). The switch-on time (Output Enable) determines the visible brightness so is used for the brightness control of each pixel to PWM.

So to answer your question: Daisy chained displays are just served by sending more data in the shift-register phase. So if you choose the `--led-chain=2` option, 64 pixels are shifted in instead of 32.

closed 2017-09-04 by hzeller