hzeller/rpi-rgb-led-matrix #390: Scan rates 32S, 8S, 16S

noorabrar1, 2017-09-06:
Hello sir,

I am new to git hub..

I'm working on HUB75 for a project, the connectivity with pi3, and i am using your client server code.
using 3 different scan rate HUB75 board each four 32x32 panel connected in series and parallel

code:

Server: sudo ./ft-server --led-parallel=2 --led-rows=64 -D64x64 --led-chain=2

Client: sudo ./send-text -g64X64+0-0+1 -h localhost "hello" -f fonts/6x12.bdf -c ff0000

    HUB75 - P5-16S-32x32-B0 consist of ABCD data lines (Works absolutely fine with series and parallel connections)

*HUB75- P2.5-32S-64X64 consist of additional E data line, wiring is properly connected(Not working properly, some set of upper led wont glow even i made correction -D128x128)
*HUB75- P5-8S-32x32 ABC data lines(Not working properly, multiple prints appear )

Please help me to set different scan rate. is there any other libraries has to downloaded or any other arguments i need to give.. ??

Sorry for posting in that issue i thought it was similar

hzeller, 2017-09-06:
What are the options you give for each of these panels ?
  - the 4 panels of 32x32, connected parallel need `--led-chain=2 --led-parallel=2`
  - the 4 panels 64x64 need `--led-chain=4 --led-parallel=2 --led-rows=64`
  - Panels with 32x32 and only ABC line need `--led-chain=4 --led-rows=16` and you have to write a transformer that matches how the panel maps the pixels.

noorabrar1, 2017-09-06:
lets consider 1 panel HUB75-P5-8S-32x32 ABC data lines

server:  sudo ./ft-server --led-chain=2 --led-rows=16 --led-parallel=2 -D32x64
client: sudo ./send-text1 -g32x64+0-15+1 -h localhost "hello" -f fonts/6x12.bdf 
 
its displaying something like this below

[image]

noorabrar1, 2017-09-06:
Thanks for info sir,
That was just 32x32    2 panel which i connected in parallel. for --led-chain=4 its not showing anything.
should i download and install transformer code???

hzeller, 2017-09-06:
You need to write a transformer.

hzeller, 2018-01-21:
If you have a wierd mapping, you might try the new option `--led-multiplexing` https://github.com/hzeller/rpi-rgb-led-matrix#multiplexing
It does the mapping built into the library.

closed 2018-01-21 by hzeller