hzeller/rpi-rgb-led-matrix #391: Brightness on chainlength != 2

pstadler1990, 2017-09-06:
I'm getting a strange result if the chainlength is set to be different from 2, i.e. if I set it to 1 and change the display's brightness (I can dynamically set it via MQTT or an attached potentiometer), everything works fine (as in the brightness changes increases/decreases correctly) until I reach something about 70%. Then the display's brightness suddenly increases to be 100% bright (at a originally 70% level), but at 71% and higher, the brightness is correct again.

hzeller, 2017-09-07:
How do you change the brightness with MQTT or the potentiometer ? This is not part of the library, so I don't know how you interface there.

pstadler1990, 2017-09-07:
I wrapped your Python binding into my display driver class. 

inside mail loop:
			```self.MYDISPLAY.clear()
			
			#run program ...
					
			self.MYDISPLAY.buffer()
					
			#Bar with ADC brightness percentage
			if self.eventPotentiometer == True:
				self.showBarTimer -= 1
				if self.showBarTimer > 0:
					self.MYDISPLAY.brightness = self.eventPotentiometerValue
				else:
					self.eventPotentiometer = False```

where self.eventPotentiometerValue is a value between 0..100.

self.MYDISPLAY.brightness calls `self.matrix.brightness = bPercentage ` from the Python wrapper.

hzeller, 2017-09-07:
mmh, that does sound strange. Did you do a basic test without the potentiometer and other distractions and just have very simple program that in a loop (pseudocode:)
   for b in 1..100 { change-brigthness(b); display something }
and see if it also behaves strangely at the 70% mark ?

How is the potentiometer connected ?

pstadler1990, 2017-09-07:
The potentiometer should not be part of the problem, as the error still occurs if I change the brightness via MQTT. Note: This issue only happens if the chain length is selected to be 1. 
I will try to create a small test case as you described and give feedback asap.

hzeller, 2018-01-21:
did you get it to work ?

closed 2018-01-21 by hzeller