hzeller/rpi-rgb-led-matrix #392: 32x32 Matrix not showing any sign of life

MrTomacco, 2017-09-06:
I'm fairly new to this topic, but still setting up that whole thing and at least starting the demo would be something I can accomplish. So I got one of those HUB75D panels and I wired it up according to your schematics (wired it to the connector saying "input"). For Power I bought a 5v 2A psu. VCC is connected to + and gnd to -. I downloaded the github code and I can even start the demo, it then says something like "press ctrl+c to end". So far so good.. if the Matrix would light up somehow, but it just doesn't, did I do something wrong with the wiring? Also I have tried out both of the power connectors my panel has.. None works. So does anyone have any idea what might be wrong? And about the wiring, at first I only connected the bottom ground to pin 9. Afterwards i tried with connecting 3 gnd on the display to 3 gnd on the rpi. Never have I seen any sign of light on the matrix. Any ideas what I could try to do?

Helbrax, 2017-09-08:
What command are you using to start the demo?
Also, are you running it as root(via sudo)?

MrTomacco, 2017-09-09:
Im using it via sudo. It even says to push ctrl+c if i want to stop the demo. On Monday my Arduino nano will arrive, so I can test it with the nano in case I unknowingly broke the rpi somehow? Else it might be a faulty led matrix perhaps.

make -C examples-api-use 
sudo examples-api-use/demo -D0

hzeller, 2017-09-09:
Also make sure to wire up things according to the [wiring diagram given in this project](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md). Some people reported that their panel was not showing anything when they used some very out-dated and wrong Adafruit tutorial for the wiring.

But yes, the Matrix boards only guarantee 5V logic and might not work with 3.3V. There are several potential issues mentioned in the [troubleshooting section](https://github.com/hzeller/rpi-rgb-led-matrix#troubleshooting).

If you have the latest Raspbian Stretch also make sure that you have a version of this library after August 19th, as that distribution changed the way how Pi versions can be detected.

MrTomacco, 2017-09-13:
From the beginning I have used your wiring diagramm, then I rewired using 2 logic level shifters and I used the library version newer than August 19th on the newest Raspian Stretch. Also checked the troubleshooting section and had audio disabled from the beginning.

Unfortunately I have not had enough time in the recent time to try out with arduino nor will I have in the next time i suppose.. Still does not work, for those who might have similiar problems, it might be because of badly soldered headers for the gpio pins, not sure though. The solder I used for the raspberry was really really bad. It did not flow very well unlike the solder I used to solder the headers for my arduino.

Once I get the time to try it out with the arduino I'll give an update. But still thanks for the help and also great library, makes everything much easier!

pinheadmz, 2017-10-04:
I am also getting zero signs of life with a [32x32 matrix](https://www.adafruit.com/product/2026). I'm using an [adafruit matrix HAT](https://www.adafruit.com/product/2345) with a Raspberry Pi 3. I just swapped in the Pi3 with a fresh installation of Raspbian Stretch, it used to be a Pi2 Model B v1.2 running Jessie. The wiring and power supply have not changed and worked great for over a year. I also pulled the latest master from https://github.com/adafruit/rpi-rgb-led-matrix to include the August 17 "pi2 detection" update. All the tests run without any console errors, but nothing lights up on the matrix... What to try next?

maddogs, 2017-10-04:
Don't use the Adafruit Library.

I'd start from fresh using the latest code from Henners repository and follow the readme for configuring and wiring using the Adafruit Hat under the section "If you have an Adafruit HAT"

pinheadmz, 2017-10-04:
Ok thanks, I got the matrix to work with this repo. Strange, I've been using the Adafruit library on projects with Pi 2 and 3, on Jessie... I guess something finally broke with Stretch?

closed 2018-01-21 by hzeller