hzeller/rpi-rgb-led-matrix #393: Distorted display

arjan-io, 2017-09-07:
Hi,
I finally have my setup together but there seems to be a problem. so far
RPI 3, install of rpi-rgb-led-matrix all went fine. Disabled audio in /boot/config.txt to be sure
Adafruit hat
HUB75 panel
5V, 4A DC powersupply

I get these distorted pictures:
[https://ibb.co/mtCLia](https://ibb.co/mtCLia)
[https://ibb.co/mCZNwv](https://ibb.co/mCZNwv)

I also tried the `-w` option set to 0, 1(everyting very flickery) and 2 to no avail.

After running a demo, I have this left [https://ibb.co/kZnWpF](https://ibb.co/kZnWpF). That same blue line sits on top of the display at startup after a power-down.
Any tips or tricks?

hzeller, 2017-09-07:
You might have an out-door panel with 1:8 multiplexing.
So you probably need to give `--led-chain=2 --led-rows=16` and then write a transformer that fixes the pixel-mapping.

arjan-io, 2017-09-07:
Hi,

Thank you for your quick response!

> So you probably need to give` --led-chain=2 --led-rows=16`

I assume the go into the /lib/Makefile? and then `make clean`, `make`

> write a transformer that fixes the pixel-mapping

I honestly have no clue as where to start ;(

What I trying to achieve in the first place, is to use 2x2 as one 'pixel'. So this might be the time to include the pixel mapping?

So far, I have this for my own code:
```
# Constants
MATRIX_W      = 32 # Number of Actual X Pixels
MATRIX_H      = 32 # Number of Actual Y Pixels
MATRIX_DEPTH  = 3  # Color Depth (RGB=3)
MATRIX_DIV    = 2  # Physical Matrix is Half of Pixel Matrix

# Enumerate RGB Matrix Object
matrix = Adafruit_RGBmatrix(MATRIX_H, MATRIX_W/MATRIX_H)
```

Best regards

hzeller, 2017-09-07:
>
> I assume the go into the /lib/Makefile? and then make clean, make
>

No, you just give these options when you start the program. There are
[options that control many aspects](
https://github.com/hzeller/rpi-rgb-led-matrix#changing-parameters-via-command-line-flags)
of the geometry etc.

write a transformer that fixes the pixel-mapping
>
> I honestly have no clue as where to start ;(
>

In the [remapping coordinates](
https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use#remapping-coordinates)
section, there is a highlevel overview.
The [header file](
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/include/transformer.h)
describes the interface and in [lib/transformer.cc](
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/transformer.cc)
you find some examples.

If you have a standard 1:8 panel, have a look in the issue tracker, some
users already did the mapping themselve (search for 1:8 or 1/8. E.g. #311
or #299.

What I trying to achieve in the first place, is to use 2x2 as one 'pixel'.
> So this might be the time to include the pixel mapping?
>

No, your pixel mapping maps single logical pixels to single pixels on the
display, not four pixels.

So far, I have this for my own code:
>
> # Constants
> MATRIX_W      = 32 # Number of Actual X Pixels
> MATRIX_H      = 32 # Number of Actual Y Pixels
> MATRIX_DEPTH  = 3  # Color Depth (RGB=3)
> MATRIX_DIV    = 2  # Physical Matrix is Half of Pixel Matrix
>
> # Enumerate RGB Matrix Object
> matrix = Adafruit_RGBmatrix(MATRIX_H, MATRIX_W/MATRIX_H)
>
>
Don't use the Adafruit code, it is hopelessly outdated (they forked my code
a long time ago) and can't even deal with the --led-chain etc. flags. Use
the code in this library.

arjan-io, 2017-09-07:
Hi Henner,

Okay, I'll get on it. I must admit that I was a little lazy and now, just before you last comment, found some issue trackers. I have a look at the ones you suggested as well.

The code I use is a copy of some else. I try to make a wordclock just like that guy. My coding skills in this direction are very limited. It will take some time to figure it out I guess.

I'll report back if at least the mapping is sorted.

Tanks.

arjan-io, 2017-09-07:
I think the whole project is a step to far for me ;(

Have been busy the whole afternoon tot write a file that does
  for (int y = 0; y < matrix->height(); ++y) {
     for (int x = 0; x < matrix->width(); ++x) {
        matrix->SetPixel(x, y, 255, 255, 255);
     }
  }
so I can check in what order they appear but to no avail using the API. Let alone implementing a transformer code. 

After hours, I ended up using Python instead and found the following:
- I need tot start at position x = -1 and y = 0 for the top left corner
- After incrementing y for the specific x, where y = 9, the LED on y = 0 lights up again too and is now running like current y - 8 at the specific x column together with what y actually is
- I still get a blue square in the top left 16x8 pixels
Python code:
```
#!/usr/bin/env python
from samplebase import SampleBase
from rgbmatrix import graphics
import time

Class test1(SampleBase):
    def __init__(self, *args, **kwargs):
        super(test1, self).__init__(*args, **kwargs)

    def run(self):
        offscreen_canvas = self.matrix.CreateFrameCanvas()
        font = graphics.Font()
        font.LoadFont("../../../fonts/4x6.bdf")
        textColor = graphics.Color(255, 255, 0)
        x = 0
        y = 0
        my_text = "."

        while True:
            offscreen_canvas.Clear()

            for x in range(-1, 32):
                for y in range(0, 32):
                    print "x=", x, " y=", y
                    offscreen_canvas.Clear()
                    graphics.DrawText(offscreen_canvas, font, x, y, textColor, my_text)
                    time.sleep(0.05)
                    offscreen_canvas = self.matrix.SwapOnVSync(offscreen_canvas)

# Main function
if __name__ == "__main__":
    test_1 = test1()
    if (not test_1.process()):
        test_1.print_help()
```

I think I need to check my soldering although it looked pretty good. If that is to no avail, I think the screen is faulty. Hard to check though what the exact cause is!

arjan-io, 2017-09-14:
Heller,

Thank you for all your support. I've acquired a new display that works perfectly fine now.

Al the best!

closed 2017-09-14 by arjan-io