hzeller/rpi-rgb-led-matrix #395: Increased brightness for outdoor use?

simonhaines, 2017-09-13:
I'm bringing up a 32x64 panel for outdoor use in direct sunlight. When comparing the same panel driven by this library on a RPi3 with the Adafruit Hat and PWM mod (using a lightweight linux with isolcpus configured) with an [Arduino UNO shield](http://embedded-lab.com/blog/rgb-matrix-driver-shield/) (admittedly only in 16x32 mode), the UNO shield appears significantly brighter.

I've experimented with the LSB nanoseconds value and reduced bit depths but without making any difference, are there any other configuration options I can try (outside command line options the demo supports) to try and boost the brightness on my 2000HZ refresh rate 1/64 scan panel?

hzeller, 2017-09-13:
Generally, I'd play with `--led-pwm-bits` as you already did and `--led-hardware-pulse`. You can also try to reduce `--led-slowdown-gpio`, but usually LED panels are not that fast.

There is a reason for lower brightness: tare some hardware limitations that prevent that we can use the full potential on-time of the LEDs, because we support the 11 bit PWM, so need to have pretty accurate timing:
  * With the default `--led-hardware-pulse` there is a dead time of about 1/8 of the time due to reset down-time for the hardware unit that generates the pulse. It does allow to keep the output enable signal on while we clock in data, so this has significant advantages with longer chains.
  * With `--led-no-hardware-pulse`, we can't parallelize clocking in data and sending output on. So it does not have the 1/8 time reset time, but we have dark time due to clocking in data (that can be reduced with less `--led-slowdown-gpio`).

Having said that, I could experiment having the latter actually maximize the on-time in the case of `--led-pwm-bits=1`. In that case (and only then), we don't need accurate timing, so the output enable signal can be left on while we are clocking. I will not have time for that before the weekend though.

Where do you get the 2000Hz refresh rate ? That is pretty high and often panels are not that fast, but I suppose you might get that refresh rate with `--led-pwm-bits=1`.

Another way to get higher brightness is to actually get outdoor panels. They typically have a different scan rate (e.g. a 32x32 panel would have 1:8 instead of 1:16). This means that te relative time the LEDs are on can be higher.

I wonder where you got the 1:64 panel ? I suspect you mean 1:32 (as a 64 pixel high panel has 1:32).

simonhaines, 2017-09-13:
Thanks Henner,

[Here is the panel](https://www.aliexpress.com/item/Outdoor-led-display-module-p5-outdoor-full-color-led-screen-module-p5-rgb-p5-led-module/32634996931.html), it is rated as outdoor and the refresh rate is listed in the specification. Sorry I did not clarify it is a 1:32 panel, I plan to chain two of these together, but am experimenting with just one at the moment.

I'll conduct some more experiments with the configuration items you mention. I don't need a high refresh-rate (I'll be displaying a series of 2- or 3-colour images for a second each) and want to maximise the on-time. Depending on the outcome, I'll look more closely at your other suggestion of leaving on the output enable signal while clocking. I'll keep this thread posted.

simonhaines, 2017-09-21:
OK, some further tweaking with bit depths and GPIO slowdown values has this panel at a decent brightness for outdoor visibility.

Thanks for your effort on this high-quality library. I have a fork with a 1/64 scan transformer if you're interested.

closed 2017-09-21 by simonhaines