hzeller/rpi-rgb-led-matrix #397: Trouble running startup script

achav2, 2017-09-18:
I'm trying to launch some matrix code from a startup script in /etc/init.d/. After a lot of trouble, I had everything working with the standard adafruit-hat option. I was still getting some occasional flickering so I then went and made the adafruit-hat-pwm mod and rebuilt. 

Now my script won't work at startup. Everything works just fine if I run the script from the console, so I'm wondering if there's some extra dependency with the -pwm option that isn't loading first?

I'm on the latest raspbian jessie lite release (as of the beginning of this week) as suggested. And the script is below.

Also, on a separate note, when I do taskset after manually starting the script I see that it's affinity is 0-2. I've added the isolcpus=3 line to cmdline.txt and see only a couple kernel processes on cpu3. Any thoughts?

#!/bin/bash
### BEGIN INIT INFO
# Provides:          testMatrixStartup
# Required-Start:    $all matrixWebServerStartup
# Required-Stop:     $remote_fs $syslog
# Default-Start:     2 3 4 5
# Default-Stop:      0 1 6
# Short-Description: Start daemon at boot time
# Description:       Enable service provided by daemon.
### END INIT INFO

sudo /myMatrix/testMatrix -f /fmyMatrix/10x20_32high.bdf --led-slowdown-gpio=2 --led-chain=14 -C255,0,0 &

hzeller, 2017-09-19:
This doesn't look like a proper init script; you need to distinguish start/stop conditions given on $1. Now, it always will start  the program, even if invoked with `stop`. So you might want to make this a more stndard init.d script. Also, you don't need `sudo` in your init script, as it is executed as root.

With regard to the PWM not working: Maybe you have something else that messes with the hardware subset the library needs, such as some audio device (make sure to disable the audio part as suggested. Also double check in /boot/config.txt that it is disabled as well). Or you have some other process that is also started in your startup-phase that messes with the initialization of that hardware subsystem. So maybe make sure to start it last in the init scripts (S99...).

If you have added isolcpus=3, then you should see the only user process being the matrix update process. You can see that in `top` if you press the key `1` to display the individual CPU load.

I would suggest to use the `--led-daemon` option to run it as a daemon.

achav2, 2017-09-20:
Thanks for the quick reply. 

I went and did a lot more research on startup scripts after reading your comment. I've switched over to the proper `systemd` method for Raspbian and it's working great with the pwm and everything. And it even restarts if the process is killed or crashes! The new unit file is below. And could probably be more well-rounded if anyone has any hot tips. It doesn't seem like it can be run with the `--led-daemon` option with `systemd`, however.

I can see with `top` that the processor load is only ever really heavy on CPU 3, but the `p` option never still never seems to show CPU 3. I did not try taskset again but I suspect it still shows the same thing.

```
[Unit]
Description=Matrix Display Service
After=matrixWebServer.service

[Service]
ExecStart=/myMatrix/testMatrix -f /myMatrix/10x20_32high.bdf --led-slowdown-gpio=2 --led-chain=14 -C255,0,0
Restart=always
RestartSec=1

[Install]
WantedBy=multi-user.target
```

closed 2018-01-21 by hzeller