hzeller/rpi-rgb-led-matrix #399: issue 64x32 panels switching to a pattern of alternate 8 rows active leds

narioinc, 2017-09-23:
Hi All,

I have a 64x32 panel which i am using to build a pixel display for my rpi3 and i tried running the hzeller code for the Panel. 
I have had experience with the code base and have also created transformers for a couple of matrices that had unusual mapping but this panel surprises me.

So when I run the demo code, with -D0, the code runs fine and the rotating square comes up perfectly (user both a single 64x32 and two in parallels with --led-parallel=2) but after sometime the matrix start showing only alternate 8 rows of leds lit up.

# When starting the script:

[image]

# After a few seconds:

[image]

And then it alternates for a few seconds between the two modes.

I am using the following hardware
1) raspberry pi 3 
3) RGB matrix board as manufactured by Electrodragon using the schematics provided in this repo. The panels are connected to the 1st and 2nd port of the board.

I have checked the following:
1) User shorter HUB75 cables to avoid any potential timing issues
2) tried setting the --led-slowdown-gpio to 0 but shows weird artifacts and hence reverted to default as it works alright.

Could someone guide me in the right direction ?

Regards
Naresh K

hzeller, 2017-09-23:
Did you check the 1wire configuration mentioned in the Troubleshooting-section?

narioinc, 2017-09-23:
Thanks for the quick response !!

Yes, the 1 wire protocol is disabled in my rpi3. I had disabled both the sdn_bcm2835 and the 1 wire after reading the section prior to using the adapter.

hzeller, 2017-09-23:
The symptom indicates, that one of the D-address line is not properly connected.
 Given that it is only happening after a couple of seconds I suspect that, if you don't have a level converter, the 3.3V logic level is just barely not enough or that the panel input is dragging down the Raspberry Pi too much.

narioinc, 2017-09-23:
Sure ill try to do level conversion quickly. 

So the level conversion should be done on the A B C and D lines alone right ?? or would the entire HUB75 port be level converted ?

Thanks once again for looking into this issue.

hzeller, 2017-09-23:
You always should have a level converter for all signals if the panel can't deal with 3.3V logic signals. It is in general a good idea as it does not strain the pi outputs.
You can just solder up the [adapter that is in this project](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3).

narioinc, 2017-09-23:
Sure

I guess I am using a small modified version of this board which also adds a eeprom and a rtc. its here 
http://www.electrodragon.com/product/rgb-matrix-panel-drive-board-raspberry-pi/

However the board design you have has a nice 5v and gnd connection for the supplying enough juice. Ill cook up your version of the board and checkout.

narioinc, 2017-09-23:
Closing the case for now as ill be testing with a level-shifted adapter board and reopen if issue persists.

closed 2017-09-23 by narioinc

hzeller, 2017-09-23:
Before you go through the hassle of soldering up a new board, maybe double check if your other board is ok, or maybe replace the HCT245.

narioinc, 2017-09-23:
Sure Henner,

Ill do a little more debugging into this. I would have love to have a scope to find out the root cause :) Anyways I can't thank you much for looking into this. I am planning to use some of these panels for a project of mine; PiClock with Alexa integration. Again completely open source with all the files (board + schematics) to be released soon. Your library has been a great time saver. 

Will update you if I get things running.

narioinc, 2017-09-23:
I had some P10 outdoor panels 32x16 that worked perfectly with version of the adapter board I have and then tried a 32x32 one before going for the 64x32 indoor ones. On my existing board ill try and hook the 5v lines (currently connected to the rpi 5v pin) connect to an external 5v supply.