hzeller/rpi-rgb-led-matrix #400: using the library for hub 12

aliyazdanipak, 2017-09-28:
there are some diffrence between hub75 and hub12
-LED switch (managed by --led-inverse)-Thank you
-OE polarity
can you explain more about customizing codes to solve the problem? i think using inverter is not a ideal way to handle the problem. 
tell me some key word , help me understand the codes
should i focus on "SendPulse" function? where do you set oe when you want to send a pulse?
 this is a great library . let's make it better.

aliyazdanipak, 2017-10-01:
after 1 day i changed OE polarity
It is the result : 
http://goodchange.ir/photo_2017-10-01_15-22-59.jpg
in this image i tried to show "I" .
what is the problem?

MaitreDede, 2017-10-01:
Hi,
For a HUB12, maybe the hzeller's lib it too powerfull. Do you need to display different levels of light per led or on/off is sufficient ?

aliyazdanipak, 2017-10-02:
hi no i dont need diffrent levels of light 
do you have any experience?

MaitreDede, 2017-10-02:
For Arduino, you can try the Freetronics [DMD](https://github.com/freetronics/DMD) lib. It is really simple to use and can be ported to RaspberryPi using Wiring Pi. It uses pre-hardcoded fonts (it can be extended).
I have ported it to C#+pigpio for my own projects, and it works fine. I have also made a higher level driver, that expose the DMD as a drawing surface, so I can use a real imaging library to draw on the panel, with a descent refresh rate.

aliyazdanipak, 2017-10-02:
Thanks for your reply
I know that library , i should work on raspberry there is no another option.
how can i port the arduino lib for raspberry?
what can id do with spi library ? are they supported by wiringpi?
#include "pins_arduino.h"
#include <avr/pgmspace.h>
#include <SPI.h>
I will start working on it tonight . but can you explain more? can you only name the parts should be changed in dmd lib?
i appricate your help.

aliyazdanipak, 2017-10-02:
also dmd2 uses arduino timer , is it possible to replace it by thread? I should work on it . you are right ! dmd library is the best option for hub12 .THANK YOU

hzeller, 2017-10-02:
You can use rpi-rgb-led-matrix to use up to 18 parallel chains of HUB12 displays (as all R1G1B1,R2G2B2 can be used as separate monochrome channel), you just need to change the OE to positive logic (either with an inverter or just in the SendPulse method; if you use the hardware pulse, you have to consult the data-sheet, the manual pulse is simple. Don't forget to flip the default bit in the main-loop) and do a little bit of display mapping to spread out the color bits to single bits.
If you don't want/need gray-scale, you can use `--led-pwm-bits=1`.

Is pretty straightforward, so I leave this as an exercise :)

NeerajShrivastava04, 2018-02-16:
Folks...has anyone got the solution for the changing the polarity for OE signal through software. I am working on HUB12 board and am not been able to change the OE polarity from software....
Can someone please help me out in this regard.

Thanks in advance for your help.

closed 2025-11-27 by marcmerlin