hzeller/rpi-rgb-led-matrix #405: Unable to open image with led-image viewer

himura95, 2017-10-06:
Hi, thank you very much for this awesome tool !

I have a little problem, I am unable to open image with led-image viewer with my Rpi1 :(
I have install libgraphicsmagick++-dev and libwebp-dev
The make don't report error.

In the console when I write : sudo ./led-image-viewer mongif.gif --lde-rows=32 --led-chain=4
He said : 
Size : 128x32
Loading 1 files...
mongif.gif skipped: Unable to open (Magick: Unable to open file (mongif.gif) reported by magick/blob.c:2834 (OpenBlob))
No image could be loaded.

But if i load the gif with ./video-viewer, the gif play nice on the matrix.

Have you an idea for make this work ?

closed 2017-10-07 by himura95

soyxan, 2017-10-25:
I have the same problem with Raspbian Strecht (in Raspbian Jessie it works).
Did you solve the problem?

himura95, 2017-10-26:
Yes but I don't know how :(
Try to make it again

bytemage, 2018-05-20:
I've prepared 90 images for my project, and two of them won't load. Same error, no matter how I save them. I tried many different combinations of the file options but they just won't be loaded by the OpenBlob code. Looks very much like it's not the file structure, but the image data that's causing this error. As strange as it sounds ...

bytemage, 2018-05-25:
It seems that error can mean anything. If you give a specific but non existing filename as parameter you get the same error. My assumption of it being the data was very likely wrong. It just didn't manage to open the file at all.

Grudz, 2019-07-02:
I am getting this same error. Perhaps it has something to do with the distribution of linux we are using (mine is also stretch)