hzeller/rpi-rgb-led-matrix #406: Info: ready made active adapter board

marcobrianza, 2017-10-07:
I have just tested this ready made active adapter board on a Pi3:
http://www.electrodragon.com/product/rgb-matrix-panel-drive-board-raspberry-pi/
It works.
I'm using it with 18 64x32 P3 panels
[image]
[image]
[image]

clgyl, 2017-10-29:
Ohh this is nice! Can I ask you where they shipped it to? I.e. what country do you live in?
Br 
Claes

marcobrianza, 2017-10-29:
Italy

maddogs, 2017-10-29:
I bought a couple too, shipped to the UK within a week. All Worked great.

clgyl, 2017-10-30:
Thanx for all the info! I have just order one myself, ship to Sweden

soyxan, 2017-11-07:
My board is on the way. But reading the product info page I have seen extra features, anyone can explain what can we use them for:

- Extra on board RTC DS1307 via IIC interface, can not be used on same time with P3 port, select by switch.
- Extra on board AT24C256 EEPROM, 256K memory on alternative IIC 25 and 26 pins. Raspberry PI 3 can NOT support alternative IIC interface, so can only be used Raspberry Pi 2.

Thanks!

maddogs, 2017-11-07:
The RTC DS1307 is Real Time Clock. I have no use or it as my pi updates Time Date via internet. May be handy if the Pi is going to be stand alone and powered off regularly.

Not sure what the EEPROM is for.

depili, 2017-11-07:
The extra stuff are of quite limited use. The RTC chip used is quite inaccurate and it will drift. EEPROM is a form of solid state storage commonly used in microcontrollers and generally inferior than using the sd-card on the pi.

somebox, 2017-11-07:
Can you tell us what kind of panels you are using with this board? Looks great!

maddogs, 2017-11-07:
I'm using what I'd call the standard 64x32 panels from Ali-express (two 32x32 internally chained) they don't need any transformer written.

marcobrianza, 2017-11-07:
this is the link to the panels I have used
https://it.aliexpress.com/item/p3-full-color-led-module-high-clear-high-resolution-black-leds-high-contrast-ratio-smd-0606/977188631.html?spm=a2g0s.13010208.99999999.289.ZyoF2k

alverman, 2017-11-26:
How match cost one for dogana?
Im from italy

Quanto hai speso per la dogana italiana?
Varia in base al quantitativo?
Ciao, Alberto

marcobrianza, 2017-11-26:
I used a shipping method with prepaid customs duties

alverman, 2017-11-27:
?
Me lo spieghi in italiano :)

marcobrianza, 2017-11-27:
ho usato  SFCQMITR: International + BRT.it (Referential ETA: 10-20 days):
e non ho pagato la dogana

alverman, 2017-12-01:
Grazie 👍

tibo2noix, 2017-12-05:
Hi,
I want to use this to control 2 different displays of 2 different size.
- 640 * 32px (10x 32*16 panels)
- 128*128 px (4x 64*64 panels)
Is it possible ?

I purchased already the active drive board from electrodragon but haven't received it yet.

Thank you very much, I am looking forward to use this amazing work you have done !

Rodriguezjp, 2022-03-21:
Hola, compre esta tarjeta hace unos meses pero no consigo ver nada en mi Panel LED, 320x160, existe alguna pagina con la que seguir todos los pasos para instalcion desde una raspberry pi zero w. ? soy un poco novato en este mundo. Gracias

Rodriguezjp, 2022-03-21:
Hello, I bought this card a few months ago but I can't see anything on my LED Panel, 320x160, there is a page with which to follow all the steps for installation from a raspberry pi zero w. ? I am a little newbie in this world. Thanks

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828