hzeller/rpi-rgb-led-matrix #411: Background Color of LEDs is White and Color Issues

kythre, 2017-10-17:
When I use text-example, the text is blue and everything else is white.
Changing the brightness only changes the brightness of the text and not all of the LEDs being white, so when the brightness is about 80, the text beings to become washed out so the whole panel becomes white.
If I change the background color to 255,0,0 the background becomes cyan regardless of the RGB sequence

hzeller, 2017-10-17:
Sounds like your panel uses inverse logic. Did you try the `--led-inverse` flag ?

kythre, 2017-10-17:
That didn't fix it but  --led-no-inverse did, thank you

closed 2017-10-17 by kythre

hzeller, 2017-10-17:
Wait, the default should be that it is not inverse. If you had to say `--led-no-inverse`, then this could only be if you had defined `INVERSE_RGB_DISPLAY_COLORS` in the Makefile. Did you ?

kythre, 2017-10-17:
I guess I must have some how. I don't remember doing anything other than what is in the readme. I re-cloned this and now it works without the flag