hzeller/rpi-rgb-led-matrix #413: led-image-viewer not working on latest Raspbian Strecht Lite

soyxan, 2017-10-24:
After installing the required packages:

`sudo apt-get install libgraphicsmagick++-dev libwebp-dev -y
`

If I try to display an animated GIF i get the following runtime error:

`Trouble loading  (Magick: No decode delegate for this image format () reported by magick/constitute.c:1529 (ReadImage))
`

The same GIF file was correctly displayed on Raspbian Jessie Lite.

How can I install the required delegates?

hzeller, 2017-10-24:
I haven't seen this error message before. But maybe there is some additional image library dependency missing ?

Did you double-check, that the image is readable by everyone ? The image viewer drops permissions to daemon/daemon, so that would need to have read access (you can switch this off with `--led-no-drop-privs` for a quick test if it is permission problems).

soyxan, 2017-10-24:
The image is readable by everyone and adding your argument to the command-line does not have any effect.
This same setup was working on Raspbian Jessie Lite.
Following the runtime error, I have found this definition of "delegate" in the GraphicsMagick webpage

`Add on third-party libraries and applications which extend the formats supported by GraphicsMagick may be obtained from this directory. Windows users can usually ignore the contents of this directory since all common delegates are included in the Windows source and binaries packages.
`

Even they provide a link to a FTP where you can download those "delegates" but I have no idea how to install them on the Raspberry Pi

[ftp://ftp.graphicsmagick.org/pub/GraphicsMagick/delegates](ftp://ftp.graphicsmagick.org/pub/GraphicsMagick/delegates)

Any idea in how to proceed?

hzeller, 2017-10-24:
I'd look if there are such delegates available as debian packets.
Also, maybe inform the package owner of that libgraphicsmagick++-dev  debian packet that it does not work as intended.

closed 2018-02-17 by hzeller