hzeller/rpi-rgb-led-matrix #414: Bizarre Pixel mapping & Multiple Pixels Set (Adafruit Matrix HAT + Rasberry Pi 3)

andrewlott, 2017-10-27:
Hey all,

I purchased a 32x32 matrix from Sparkfun (https://www.sparkfun.com/products/12584) and am trying to get it working with a Raspberry Pi 3 (Jessie OS) and Adafruit RGB Matrix HAT (https://learn.adafruit.com/adafruit-rgb-matrix-plus-real-time-clock-hat-for-raspberry-pi/overview). Everything is hooked up, but the pixel mapping is very strange and multiple pixels are set at the same time. I've scoured the other issues, tried various flag combinations, and have written code for a Transformer, but I think the issue is outside of the realm of Transformers since multiple pixels are set at the same time. 

I took a video of the panel executing [Demo 1](https://www.dropbox.com/s/szk9mtk76vbs8ie/Video%20Oct%2026%2C%207%2013%2057%20PM.mov) and some [test code](https://www.dropbox.com/s/6mlxkd4jglhpdrn/Video%20Oct%2026%2C%207%2012%2053%20PM.mov) that was mentioned in another thread which goes through each X/Y coordinate and turns the pixel white . Notice that not only are multiple pixels being turned on at the same time, but that they are hit twice! I set the brightness of the pixel to be 128 so that we can see that there's a second pass, as the pixels become brighter. As far as flags, I'm only using --led-gpio-mapping=adafruit-hat for this. When I change the led-chain number to 2 or 4, it seems multiple pixels are not set at the same time, but the mapping is still strange and pixels are set twice.

Seeking any advice for how to proceed. Thanks for any help :)

KentWalker, 2017-10-27:
Hi Andrew,

I managed to get 32x32 panels working, here's what has worked for me

In the framebuffer.cc file change line 55 from:
    `double_rows_(rows / **2**), row_mask_(double_rows_ - 1) {`
to:
    `double_rows_(rows / **4**), row_mask_(double_rows_ - 1) {`

In your transformer.cc file change your TransformerXXX::TransformCanvas::SetPixel(...) function:
```
  int new_x;
  int new_y;
  new_x = ((x / 16) * 32) + (x % 16);
  if ((y & 8) == 0) {
    new_x += 16;
  }
  new_y = ((y / 16) * 16) + (y % 8);	
  delegatee_->SetPixel(new_x, new_y, red, green, blue);
```
and also change your TransformerXXX::TransformCanvas::width()
`  return delegatee_->width() / 2; `

Finally in your main file set rows = 32 and chain = (2 * number of panels in chain)

Hope this is of some use

andrewlott, 2017-11-01:
Hi Kent,

Thanks so much for the quick reply! I added your suggested changes and one side of the matrix is working perfectly! The other side is still positioned wrong, so I've been playing with the Transformer code, getting mixed results. I have videos of the matrix running your code on the [test demo](https://www.dropbox.com/s/mzho6wkmrtcmkbg/Video%20Oct%2031%2C%205%2035%2059%20PM.mov?dl=0) and [Demo 0 (the square)](https://www.dropbox.com/s/m8fcf3brtexdw1u/Video%20Oct%2031%2C%205%2036%2054%20PM.mov), as well as with my modifications (that had the best results) on the [test demo](https://www.dropbox.com/s/takl4c5orgneb5b/Video%20Oct%2031%2C%205%2048%2025%20PM.mov) and [Demo 0 (the square)](https://www.dropbox.com/s/2rgszpii5wlhq28/Video%20Oct%2031%2C%205%2042%2041%20PM.mov). My modifications were the following:
```C
  int new_x;
  int new_y;
  new_x = ((x / 16) * 32) + (x % 16);
  if ((y & 8) == 0) {
    new_x += 16;
  }
  new_y = ((y / 16) * 16) + (y % 8);	

  /* The following if-block is the change */
  if ((x & 16) == 0) {
    new_y = y;
  }
  delegatee_->SetPixel(new_x, new_y, red, green, blue);
```

At first glance the results appear to just be offset by 8px, however in the Demo 0 video it is clear that in addition to the offset there is a quadrant that is still receiving excess pixel data. Without my modifications, the quadrants in that first half are just placed incorrectly (first/second swapped and third/fourth swapped). I've been unable to solve these issues, so please let me know if you (or anyone else) have any ideas. Thanks again for your help!

KentWalker, 2017-11-01:
Hi Andrew,

I'm assuming from your videos that the module is upside down? (ie (0,0) is at bottom right corner?)

The new_x transform is ok and the new_y transform is also ok providing ((x & 16) == 1) ie x >15.
When x <16 the odd blocks ((y & 8) == 1) are being swapped with the even blocks ((y & 8) == 0).

To prove this a quick test would be:

```
  int new_x;
  int new_y;
   
  new_x = ((x / 16) * 32) + (x % 16);
  if ((y & 8) == 0) {
    new_x += 16;
  }

  new_y = ((y / 16) * 16) + (y % 8);	
  if ((x & 16) == 0) {
    if (y & 8) == 0) {
      new_y -= 8;
    } else {
      new_y += 8;
    }
  }

  delegatee_->SetPixel(new_x, new_y, red, green, blue);
```

andrewlott, 2017-11-01:
Hi Kent,

Thanks for the speedy reply again, and you're correct that the matrix is upside down in the videos. That was my initial thought as well, that the odd and even blocks are simply swapped, but when I'd tried something like your changes the results were unexpected. I put in your exact changes and these were the results on the [test code](https://www.dropbox.com/s/p5cpnxdh2vk5vfo/Video%20Oct%2031%2C%208%2038%2026%20PM.mov) and [Demo 0](https://www.dropbox.com/s/bf0wyzedppofkda/Video%20Oct%2031%2C%208%2039%2034%20PM.mov). The mapping does not seem straightforward and might depend on other factors..

KentWalker, 2017-11-01:
That is weird.  I suggest you try this...
SInce the new_x transform is working, leave the x transform but don't transform y at all.
Do a for/next loop where x= 0..15 and y = x first and see which pixels are lit so you can tell what values of y are lighting which row eg if column 7 is lit, y = 7.
Then repeat for y = x + 8, 16, 24, 32... until you manage to have lit a pixel in every row.
If you can't do this I would suspect that you may have a faulty panel.

andrewlott, 2018-01-07:
Hi Kent,

Thanks for all your help, sorry for the silence on my end. I was unable to consistently light each pixel, so I reached out to the seller of the panels and told them that I believed they were be faulty. They issued me a refund and I purchased new panels elsewhere. The other panels were definitely faulty as I am able to run the demos as expected with the new panels! Thanks for all of your help, I'll be closing this issue. :)

closed 2018-01-07 by andrewlott