hzeller/rpi-rgb-led-matrix #416: CPU usage: chained vs parallel

soyxan, 2017-10-27:
I have a setup with a 2 rows of 12 32x32 panels each using a U-shape (1 physical row).

The panels bright ok but the RPi 3 CPU usage is 150% during some transitions. 

My question is if the CPU usage will be lower if connecting the panels using 2 physical rows (no U-shape)

hzeller, 2017-10-28:
Yes, this will reduce CPU and improve refresh rate (=less flicker).

Sending things in parallel does use the same CPU per column, but now you can do twice as many columns per clock. If anything, you'll see that the refresh rate goes up and you have less flicker.

In your case, I'd even suggest to make three columns each with 8 panels in a row, but then arrange them in a way that you still have your 2x12 arrangement, and write a transformer to map all the pixels to the right spots.

soyxan, 2017-10-30:
Ok I will try.

On the other hand will isolcpus=3 have any effect on this high CPU usage?

hzeller, 2017-10-30:
isolcpus=3 will only make sure that the update thread is on one core and limit possible flickering.

The update thread will at most take 100% CPU, everything else is due to the program generating the pixels; so if you see 150% CPU, at least 50% is by your program.

soyxan, 2017-10-30:
I am going to ask this in another way :)

My program is doing some image manipulation (with Magick) before displaying each frame in order to make some effects (rotations, fades, zooms, scrolls, etc.)

Is there any situation in which isolcpus=3 will affect negatively the performance of the pannels?

hzeller, 2017-10-30:
isolspus=3 will reserve one core for the panels to update wihtout being interrupted by something else. So it will strictly improve the performance of the panels. Also with your current length of the panel, I also suggest to experiment with the `DISABLE_RT_THROTTLE` option in the lib/Makefile.

There are 3 cores left to do the rest. Which is totally fine in your case, as you don't use that much more CPU with the rest. 

Be aware that doing heavy processing can still influence the output: if your image processing is memory intensive or does a lot of IO, it still will influence the output. So doing as much as possible preprocessing is always a good idea.

soyxan, 2017-10-30:
What this DISABLE_RT_THROTTLE option does?
I have read the comment on gpio.cc but I do not get it ;)

hzeller, 2017-11-02:
Usually, the kernel tries to reserve a little fraction of time that it interrupts realtime processes to avoid them hogging the system. This 'realtime throttling' can get into the way if that process actually needs that CPU (e.g. if you have a very long chain). If that happens, the symptom is, that there is a slight regluar dimming of the screen every second or so.

The `DISABLE_RT_THROTTLE` option disables that kernel setting.

closed 2017-11-02 by hzeller