hzeller/rpi-rgb-led-matrix #417: Are 120*90 30scan panels supported?

davivid, 2017-10-30:
Hi,
Does the library have support for other resolution panels such as:
120*90 30scan

davivid, 2017-10-31:
I'd like to try out some of these:
[](https://www.aliexpress.com/store/product/Normal-new-product-P1-667-indoor-smallest-pitch-full-color-led-module-360000dots-sqm-120-90mm/1186165_32827997183.html)
I think they have the same HUB-75 interface, but not sure how complicated it would be to adapt the code?

MaitreDede, 2017-11-15:
Hi @davivid , do you have such panel ?

If it is a 1/32s panel, it is driveable. The current code does not have yet the releavant bits, but I am driving one 1/32s 64x64 panel.

hzeller, 2018-02-17:
I think there is only one way to find out: Get one of these panels, connect it to the library and see what happens, possibly modify the code as needed. Please send a pull request with changes if needed.

closed 2018-02-17 by hzeller